tttapa/Control-Surface #945: Arduino UNO

KinoLab07, 2023-08-15:
Hi,
I've an arduino UNO that is hacked with hiduino and is recognize now as MIDI controller...

Did anyone have use this board with rotary encoder... I'm quite loose on how configure it with a mcp23017.

Thanks

tttapa, 2023-08-15:
Yes, the Arduino UNO can be used with rotary encoders, and this is supported by Control Surface, both with and without an MCP23017.  
Keep in mind that performance may be poor without interrupt pins (of which the UNO only has 2).

KinoLab07, 2023-08-16:
Ok, I know the poor performance of the UNO, but is the one that I've for testing for the moment... The hole project is with 64 rotary, and for that I'll buy a teensy or a due.. Anyway...

My problem is that I'm Trying this exemple: https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/dac/MCP23017-RotaryEncoder-No-Interrupts_8ino-example.html

I've conected the rotary encoder to the GPA0 and GPA1 pin of the mcp23017 and the INT A to the pin12 (I've also try to connect to pin3 ) of the arduino and nothing happen.

The SDA and SCL to pin A4 and A5 of arduino... and nothing happen.. 

Sooo I don't know where is the problem...

tttapa, 2023-08-17:
Did you connect the rest of the pins correctly? Reset/enable, address pins, decoupling caps ...

AriESQ, 2025-02-08:
Please close this issue.