tttapa/Control-Surface #948: Compilation error: expected template-name before '<' token

hiegdk, 2023-08-25:
Hi, this might be a stupid issue as I'm a Arduino noob. I'm getting the following error while attempting to compile the MIDI-Input-Fine-Grained-All-Callbacks.ino example code from the Doxygen site. 

`
/Users/REDACTEDMIDI_CC_Input/MIDI_CC_Input.ino:7:52: error: expected template-name before '<' token
 struct MyMIDI_Callbacks : FineGrainedMIDI_Callbacks<MyMIDI_Callbacks> {
                                                    ^
/Users/REDACTED/MIDI_CC_Input/MIDI_CC_Input/MIDI_CC_Input.ino:7:52: error: expected '{' before '<' token
/Users/REDACTED/MIDI_CC_Input/MIDI_CC_Input/MIDI_CC_Input.ino:7:52: error: expected unqualified-id before '<' token

exit status 1

Compilation error: expected template-name before '<' token
`

I'm using Arduino IDE 2.1.1, and an Arduino Due board. 

Is that example code even expected to work with the latest version of the library? If not, could someone point me in the right direction (I just want to read MIDI CC messages and then do other, non-MIDI related stuff based on the value)? Thanks!

tttapa, 2023-08-25:
The documentation you were using is probably for the `main` branch of Control Surface. (E.g. this page: https://tttapa.github.io/Control-Surface-doc/Doxygen/da/d27/MIDI-Input-Fine-Grained-All-Callbacks_8ino-example.html)  
The version is listed in the top right corner.

I suspect that you do not have the `main` version installed, previous versions did not yet support `FineGrainedMIDI_Callbacks`. You'll have to go to the home page of this repository and download the ZIP file using the green <kbd><> Code</kbd> button. Direct link: https://github.com/tttapa/Control-Surface/archive/refs/heads/main.zip

hiegdk, 2023-08-25:
That appears to have been it! Thanks! For future reference, I was using v 1.2.0 which I had downloaded from Releases as it was marked as latest.

closed 2023-08-25 by hiegdk