tttapa/Control-Surface #951: Arduino UNO R4 support and testing

tttapa, 2023-09-03:
The latest `main` version of Control Surface adds preliminary Arduino UNO R4 support (#950).

At this point, nothing has been tested on the actual hardware yet (I don't own an R4), but all examples compile without issues.  **If you have access to an Arduino UNO R4, please try out some of the examples, and report back the results in this thread.**

In particular:
 - The encoder library required some workarounds (e20e32fbea88535fc92baa6ea914180f97b1f2a4), which need to be thoroughly tested. Information about the `attachInterrupt` function on the R4 is scarce, and it is unclear to me what happens if a user attaches an interrupt on two different pins that happen to share an IRQ channel. The official Arduino documentation has not yet been updated for the R4, and the product spec sheet only mentions external interrupts on pins 2 and 3, which does not match the Renesas RA4M datasheet. Unfortunately, I don't have enough free time at the moment to really dig into [ArduinoCore-renesas/cores/arduino/Interrupts.cpp](https://github.com/arduino/ArduinoCore-renesas/blob/main/cores/arduino/Interrupts.cpp).
 - USB MIDI support has not yet been added. We need to wait for ArduinoCore-renesas to implement the low-level USB descriptors and APIs. Once the Core supports it, adding support to Control Surface should be trivial (see the [Custom-USB-MIDI-Backend.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d5f/Custom-USB-MIDI-Backend_8ino-example.html) example).

LukeThomasDawson, 2023-09-07:
On it!

Got an UNO R4 WiFi arriving in the post next week, I'll report back after testing the examples.

Cheers Pieter, keep up the top work.

LukeThomasDawson, 2023-09-13:
### @tttapa  So an Uno R4 WiFi just landed in the letterbox 30 mins ago...

Updating to the latest release with no uppercase CHANNEL_X constants ([aaf6eea](https://github.com/tttapa/Control-Surface/commit/aaf6eea2206c6e3b6a2004e0520100342945983f)) was needed, but otherwise everything runs fine.

---

> * The encoder library required some workarounds ([e20e32f](https://github.com/tttapa/Control-Surface/commit/e20e32fbea88535fc92baa6ea914180f97b1f2a4)), which need to be thoroughly tested. Information about the `attachInterrupt` function on the R4 is scarce, and it is unclear to me what happens if a user attaches an interrupt on two different pins that happen to share an IRQ channel. The official Arduino documentation has not yet been updated for the R4, and the product spec sheet only mentions external interrupts on pins 2 and 3, which does not match the Renesas RA4M datasheet. Unfortunately, I don't have enough free time at the moment to really dig into [ArduinoCore-renesas/cores/arduino/Interrupts.cpp](https://github.com/arduino/ArduinoCore-renesas/blob/main/cores/arduino/Interrupts.cpp).

_**Looks like we're good to go on all digital pins 👍**_ 

I tested [CCRotaryEncoder](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d62/classCCRotaryEncoder.html) on all combinations of digital inputs, and all run as expected.  The [e20e32f](https://github.com/tttapa/Control-Surface/commit/e20e32fbea88535fc92baa6ea914180f97b1f2a4) workaround does the job.

---

> * USB MIDI support has not yet been added. We need to wait for ArduinoCore-renesas to implement the low-level USB descriptors and APIs. Once the Core supports it, adding support to Control Surface [...]

Indeed, no USB MIDI support as yet... and on the R4 WiFi variant there could be some quirks with the interplay between the RA4M1 and the ESP32-S3, but I'm sure there will be some documentation out when the time is right : ) 

---

Now back to writing more MIDI output classes for my Teensy 4.1 controller...  support enabled now for the classic iPod Touch Wheel as a MIDI output object for DAW transport control 😉

Thanks again Pieter for the rock solid work.

tttapa, 2023-09-13:
Thanks for taking the time to try this out!