tttapa/Control-Surface #981: Switching presets Roland FP-30X

Wallirec, 2023-12-04:
`Hello! Please tell me how to use your library and a two-button controller to switch piano presets, there are not many of them, 56 in total. I couldn’t find examples to transfer |MSB|LSB|PC

Piano
No.|Name |MSB|LSB|PC
--------------------------------------------
1 |Concert Piano | 0| 68| 1
2 |Ballad Piano | 16| 67| 1
3 |Mellow Piano | 4| 64| 1
4 |Bright Piano | 8| 66| 2
5 |Upright Piano | 16| 64| 1
6 |Mellow Upright | 1| 65| 1
7 |Bright Upright | 1| 66| 1
8 |Rock Piano | 8| 64| 3
9 |Ragtime Piano | 0| 64| 4
10 |Magical Piano | 47| 65| 3
11 |Harpsichord | 0| 67| 7
12 |Harpsi 8'+4' | 8| 67| 7
E.Piano
No.|Name|MSB|LSB|PC
--------------------------------------------
1 |1976SuitCase | 8| 71| 5
2 |Wurly 200 | 25| 64| 5
3 |Phase EP Mix | 8| 68| 5
4 |80's FM EP | 0| 68| 6
5 |Clav. |121| 0| 8
6 |Vibraphone |121| 0| 12
7 |Celesta |121| 0| 9
8 |B.Organ Slow | 1| 65| 19
9 |Combo Jz.Org | 0| 70| 19
10 |Ballad Organ | 0| 69| 19
11 |Gospel Spin | 0| 71| 17
12 |Full Stops | 0| 69| 17
13 |Mellow Bars | 32| 68| 17
14 |Lower Organ | 0| 66| 17
15 |Light Organ | 32| 69| 17
16 |Pipe Organ | 8| 70| 20
17 |Nason Flt 8' | 16| 66| 20
18 |ChurchOrgan1 | 0| 66| 20
19 |ChurchOrgan2 | 8| 69| 20
20 |Accordion |121| 0| 22
Other
No.|Name|MSB|LSB|PC
--------------------------------------------
1 |Epic Strings | 1| 67| 49
2 |Rich Strings | 0| 71| 50
3 |SymphonicStr1 | 1| 67| 50
4 |SymphonicStr2 | 1| 65| 50
5 |Orchestra | 8| 66| 49
6 |String Trio | 0| 64| 41
7 |Harpiness | 0| 70| 47
8 |OrchestraBrs | 1| 66| 61
9 |Super SynPad | 1| 71| 90
10 |Choir Aahs 1 | 8| 71| 53
11 |Choir Aahs 2 | 8| 72| 53
12 |D50 StackPad | 1| 64| 89
13 |JP8 Strings | 0| 68| 51
14 |Soft Pad | 0| 64| 90
15 |Solina | 0| 66| 51
16 |Super Saw | 8| 67| 82
17 |Trancy Synth | 1| 65| 91
18 |Flip Pad | 1| 64| 91
19 |Jazz Scat | 0| 65| 55
20 |Comp'd JBass | 0| 66| 34
21 |Nylon-str.Gt |121| 0| 25
22 |Steel-str.Gt |121| 0| 26
10
23 |AcousticBass |121| 0| 33
24 |A.Bass+Cymbl | 0| 66| 33`

tttapa, 2023-12-11:
You could use something like this:

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

void setup() {
  Control_Surface.begin();
  // Example: select “Ballad piano”
  Control_Surface.sendControlChange(MIDI_CC::Bank_Select, 16);
  Control_Surface.sendControlChange(MIDI_CC::Bank_Select_LSB, 67);
  Control_Surface.sendProgramChange(1); // Or zero, depends on Roland's convention
}

void loop() {
  Control_Surface.loop();
}
```

Wallirec, 2023-12-12:
Thank you, I was able to assign switching to 4 banks, but I don’t understand how to switch presets in a circle with two buttons, please help.

`#include <Control_Surface.h> // Include the Control Surface library

USBMIDI_Interface midi;

IncrementButton btn1 {2}; 
IncrementButton btn2 {4};

unsigned char btn_flg = 0;
unsigned char btn2_flg = 0;
unsigned char btn3_flg = 0;
unsigned char btn4_flg = 0;

#define INFLED_PIN 14
#define INFLED_TIME 150
#define REDLED_PIN 15
#define GREENLED_PIN 16
#define ADCVOL_MIN 0
#define ADCVOL_MAX   910
#define ADCVOL_TRHLD 1018

unsigned char infled_flag =0;
unsigned long infled_timer =0;
unsigned long mil=0;
unsigned long adc_time=0;
int adc=0;
int adc_old=0;
int adc_raw=0;
#define FILTERN 5
int adc_arr[FILTERN]={0};
int adcFilter(int adc, int *arr, int n){
  unsigned int result=0;
  for(int i=1; i<n; i++){
    arr[i-1]=arr[i];
  }
  arr[n-1]=adc;
  for(int i=0; i<n; i++){
    result+=arr[i];
  }
  result/=n;
  return (int)result;
}
unsigned char adc_switch=0;
unsigned char volume_mod=0;
unsigned char volume=0;
unsigned char volume_old=0;

void setup() {
  pinMode(GREENLED_PIN,1);
  pinMode(REDLED_PIN,1);
  pinMode(INFLED_PIN,1);
  digitalWrite(GREENLED_PIN,volume_mod);
  digitalWrite(REDLED_PIN,!volume_mod);

  Control_Surface.begin();
  btn1.begin(); 
  btn2.begin(); 

}

void loop() {
  Control_Surface.loop();
  mil=millis();
  if(mil-adc_time>2){
    adc_time=mil;
    adc_raw=analogRead(A0);
    adc=adcFilter(adc_raw, adc_arr, FILTERN);
  }
  if(abs(adc_old-adc)>1){
    adc_old=adc;
    if(adc<ADCVOL_MIN){adc=ADCVOL_MIN;}
    if(adc>ADCVOL_TRHLD){
      if(!adc_switch){
        adc_switch=1;
        volume_mod=!volume_mod;
        Control_Surface.sendCC(20, volume_mod * 127);
        infled_flag=1;
        digitalWrite(GREENLED_PIN,volume_mod);
        digitalWrite(REDLED_PIN,!volume_mod);
      }
    }else{
      if(adc>ADCVOL_MAX){adc=ADCVOL_MAX;}
      volume=map(adc,ADCVOL_MIN,ADCVOL_MAX,0,127);
      if(volume_old!=volume){
        volume_old=volume;
        if(!volume_mod){
          Control_Surface.sendCC(7, volume);
          infled_flag=1;
        }else{
          Control_Surface.sendCC(11, volume);
          infled_flag=1;
        }
      }
    }
    if(adc_switch && adc<ADCVOL_MAX){adc_switch=0;}
  }
 
  //behavior for the first button
  switch (btn1.update()) {
    case IncrementButton::ReleasedShort:
      Control_Surface.sendControlChange(MIDI_CC::Bank_Select, 0);
  Control_Surface.sendControlChange(MIDI_CC::Bank_Select_LSB, 68);
  Control_Surface.sendProgramChange(1);
      infled_flag=1;
      break;
   case IncrementButton::IncrementLong:
     Control_Surface.sendControlChange(MIDI_CC::Bank_Select, 8);
  Control_Surface.sendControlChange(MIDI_CC::Bank_Select_LSB, 71);
  Control_Surface.sendProgramChange(5);
      infled_flag=1;
      break;
  }

  //behavior for the second button
  switch (btn2.update()) {
    case IncrementButton::ReleasedShort:
      Control_Surface.sendControlChange(MIDI_CC::Bank_Select, 121);
  Control_Surface.sendControlChange(MIDI_CC::Bank_Select_LSB, 0);
  Control_Surface.sendProgramChange(33);
      infled_flag=1;
      break;
    case IncrementButton::IncrementLong:
     Control_Surface.sendControlChange(MIDI_CC::Bank_Select, 121);
  Control_Surface.sendControlChange(MIDI_CC::Bank_Select_LSB, 0);
  Control_Surface.sendProgramChange(22);
      infled_flag=1;
      break;
  }
  if(infled_flag){
    infled_flag=0;
    infled_timer=mil;
    digitalWrite(INFLED_PIN,1);
  }
  if(mil-infled_timer>INFLED_TIME){
    digitalWrite(INFLED_PIN,0);
  }
}`

tttapa, 2023-12-14:
So you want to cycle through the 56 presets using two buttons?  
You'll have to use an integer variable that you update whenever a button is pressed: one button increments the variable (or resets it to zero after hitting the maximum setting), the other decrements it (or resets it to the maximum setting when going below zero). Then you look up the corresponding bank and program number in an array, using your variable as index.

Wallirec, 2023-12-14:
I think I need to switch somewhere between ten selected presets.

closed 2024-06-08 by Wallirec