tttapa/Control-Surface #984: Error running on M1 Mac Sonoma with Teensy 4.0

NickCulbertson, 2023-12-15:
Hello! I'm having an issue compiling the latest build of Control Surface on M1 Mac Sonoma with Teensy 4.0.

This is the script I'm using to test:
```
#include <Control_Surface.h>

USBMIDI_Interface midi;

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop(); 
}
```

If I run this on Teensy 4.0 with the Control Surface release version 1.2.0 it compiles.  However, if I try to compile using the main repo branch I get this error (not the full error message):
```
In file included from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Interfaces/MIDI_Pipes.hpp:11:0,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Interfaces/MIDI_Interface.hpp:3,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Control_Surface/Control_Surface_Class.hpp:10,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Control_Surface.h:16,
                 from /Users/nickculbertson/Documents/Arduino/MIDITeensy4/MIDITeensy4.ino:1:
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:58:5: error: enumerator value 256 is outside the range of underlying type 'uint8_t {aka unsigned char}'
     NONE CS_DEPREC("Use None instead") = None,
     ^
In file included from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/AH/PrintStream/PrintStream.hpp:129:0,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/AH/Debug/Debug.hpp:8,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/AH/Containers/LinkedList.hpp:9,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/AH/Containers/Updatable.hpp:10,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Control_Surface/Control_Surface_Class.hpp:5,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Control_Surface.h:16,
                 from /Users/nickculbertson/Documents/Arduino/MIDITeensy4/MIDITeensy4.ino:1:
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Settings/NamespaceSettings.hpp:19:24: error: expected '}' before '[' token
 #define CS_DEPREC(msg) [[deprecated(msg)]]
                        ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:58:10: note: in expansion of macro 'CS_DEPREC'
     NONE CS_DEPREC("Use None instead") = None,
          ^
In file included from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Interfaces/MIDI_Pipes.hpp:11:0,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Interfaces/MIDI_Interface.hpp:3,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Control_Surface/Control_Surface_Class.hpp:10,
                 from /Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Control_Surface.h:16,
                 from /Users/nickculbertson/Documents/Arduino/MIDITeensy4/MIDITeensy4.ino:1:
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:58:40: error: expected unqualified-id before '=' token
     NONE CS_DEPREC("Use None instead") = None,
                                        ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:116:17: error: 'Cable' has not been declared
                 Cable cable = Cable_1)
                 ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:120:33: error: expected ')' before 'header'
     MIDIMessage(MIDIMessageType header, uint8_t data1, uint8_t data2,
                                 ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:128:5: error: 'Cable' does not name a type
     Cable cable; ///< USB MIDI cable number;
     ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:139:5: error: 'MIDIMessageType' does not name a type
     MIDIMessageType getMessageType() const {
     ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:149:25: error: 'MIDIMessageType' has not been declared
     void setMessageType(MIDIMessageType type) {
                         ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:163:5: error: 'Cable' does not name a type
     Cable getCable() const { return cable; }
     ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:317:45: error: 'Cable' has not been declared
     SysExMessage(const uint8_t (&array)[N], Cable cable = Cable_1)
                                             ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:323:5: error: 'Cable' does not name a type
     Cable cable;
     ^
...
...
...
/Users/nickculbertson/Documents/Arduino/MIDITeensy4/MIDITeensy4.ino:1:
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp: At global scope:
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/Settings/NamespaceSettings.hpp:10:26: error: expected declaration before '}' token
 #define END_CS_NAMESPACE }
                          ^
/Users/nickculbertson/Documents/Arduino/libraries/Control-Surface/src/MIDI_Parsers/MIDI_MessageTypes.hpp:417:1: note: in expansion of macro 'END_CS_NAMESPACE'
 END_CS_NAMESPACE
 ^
Error compiling for board Teensy 4.0.
```

I see the several messages about `cable` but I'm not sure if this something that has changed in the library on in Apple's C compiler. Please advise.

tttapa, 2023-12-15:
Thanks for the report. Which version of the Teensy core are you using? I can't seem to reproduce the issue using version 1.58: https://github.com/tttapa/Control-Surface/actions/runs/7215496844/job/19659863221

NickCulbertson, 2023-12-16:
Oh, I'll try updating to that. I'm currently on Arduino 1.8.19 / Teensyduino 1.56. Thanks!

tttapa, 2023-12-16:
Looks like GCC 5 does not correctly parse C++14 attributes. I've pushed a workaround that seems to work for Teensy 1.56 as well (at least on Linux).

NickCulbertson, 2023-12-16:
Thanks for looking into it! Updating the Android IDE and Teensy version fixed the issue.

closed 2023-12-16 by NickCulbertson