tttapa/Control-Surface #985: Arduino Nano 33 BLE Support?

tilllt, 2023-12-31:
I stumbled upon this:

https://github.com/lathoub/Arduino-BLE-MIDI?tab=readme-ov-file

It seems the work to implement BLE Midi on Nano 33 BLE is done for other midi libraries... will that make it more probable you could implement Midi over BLE for the Nano 33 as well? I would like to use your library for a project but the lack of BLE Midi for the Nano 33 is prohibiting this ATM.

tttapa, 2023-12-31:
You can use transports for the MIDI library using Control Surface by using the included [`FortySevenEffectsMIDI_Interface`](https://tttapa.github.io/Control-Surface-doc/Doxygen/dd/dd7/classFortySevenEffectsMIDI__Interface.html) adapter.

```cpp
#include <BLEMIDI_Transport.h>
#include <hardware/BLEMIDI_ArduinoBLE.h>

#include <Control_Surface.h>
#include <MIDI_Interfaces/Wrappers/FortySevenEffects.hpp>

// Create an Arduino BLE MIDI instance
BLEMIDI_CREATE_DEFAULT_INSTANCE()
// Then wrap it in a Control Surface-compatible MIDI interface
FortySevenEffectsMIDI_Interface<decltype(MIDI) &> cs_midi = MIDI;

// Also route the incoming MIDI messages to the Serial Monitor for debugging
USBDebugMIDI_Interface debug_midi;
BidirectionalMIDI_Pipe mpipe;
 
void setup() {
  cs_midi.setAsDefault();
  Control_Surface.begin();
  debug_midi | mpipe | cs_midi;
}

void loop() {
  Control_Surface.loop();
  static AH::Timer<millis> timer{250};
  if (timer)
    Control_Surface.sendControlChange(7, (millis() / 250) % 128);
}
```

Unfortunately, you'll find that the library you linked to does not support the Arduino Nano 33 BLE either at this point: https://github.com/lathoub/Arduino-BLE-MIDI/issues/55  

None of the recent stable releases worked for me, but the `Arduino-Nano-33-BLE` branch does at least seem to compile.  
However, the examples do not work, as I'm unable to connect to the device after uploading (Neither on Ubuntu 22.04 with BlueZ 5.64, nor on Android 14).

I have some other concerns about the way Arduino-BLE-MIDI is implemented, and the API exposed by the ArduinoBLE library is rather limited. I'll need to find some time to actually look at this topic in detail to see if these concerns are well-founded.

tilllt, 2023-12-31:
Thanks for looking into it and happy new year :)

tttapa, 2024-01-23:
MIDI over BLE is now supported on the Arduino Nano 33 BLE, Raspberry Pi Pico, and other boards: #993

AriESQ, 2025-02-08:
Please close issue.