tttapa/Control-Surface #989: Select bank

1tumusic, 2024-01-07:
Hello Pieter P,

First, thank you for your great library. 

I need some help:

I want to use 3 buttons for different bank such as A for bank 1, B for bank 2 and C for bank 3.
I only see 2 buttons used in your example. 
So I don't understand how can I use 3 buttons for select bank.

Thanks for your help!
[image]

tttapa, 2024-01-07:
See the [Selectors](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html) documentation. You probably want the `ManyButtonsSelector`.

1tumusic, 2024-01-08:
> See the [Selectors](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d0c/group__Selectors.html) documentation. You probably want the `ManyButtonsSelector`.

Thank you!