tttapa/Control-Surface #996: Arduino Giga R1 Wifi support

BenRauzi, 2024-01-27:
Are there any plans to support the Arduino Giga R1 Wifi in future with MIDI over USB? Understandably it's STM32 so would be a bit more work, but keen to build something with this board.

tttapa, 2024-01-27:
The code for the Giga should be there, but I don't have the hardware to test it. I've just verified that the correct backend is selected and that the code compiles without warnings or errors, but I have no idea if it will work. If the GIGA USB implementation and API in the ArduinoCore-mbed are similar enough to the ones for the Nano 33 BLE, it may work without too many modifications.

To be honest, the low-level MIDI USB code for the mbed platform is rather experimental, it needs some refactoring and simplifications, along with more rigorous tests, but there have been a lot of other things on my todo list, unfortunately.

If you have a Giga at hand, it would be great if you could give it a try, it should be as simple as using the `USBMIDI_Interface`.  
I usually start with a simple loopback test:

```cpp
#include <Control_Surface.h>
// Make sure we're not silently falling back to MIDI over Serial
#ifdef CS_USB_MIDI_NOT_SUPPORTED
#error "Actual MIDI USB backend required"
#endif

USBMIDI_Interface midi;
MIDI_Pipe p;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  midi >> p >> midi;
  MIDI_Interface::beginAll();
}

void loop() {
  static bool ledState = false;
  static AH::Timer<millis> timer {500};
  if (timer)
    digitalWrite(LED_BUILTIN, ledState = !ledState);
  MIDI_Interface::updateAll();
}
```

BenRauzi, 2024-01-28:
Sure thing I'll have an experiment and see if I can get anything working

intini, 2024-07-19:
@BenRauzi Hi! Please, may I ask if you were you able to test the Arduino Giga R1 Wifi with the Control Surface library?
Has anyone? I just bought a chinese version of the Arduino Due. I hope that it will work fine with the Control Surface Library. But I am looking forward to exploring a more modern board like the new Giga R1 for this application.
I would like to thank Pieter (@tttapa) for working on such a fantastic Library. Thank you, Pieter!

agdl, 2025-02-08:
In interrupt_pins.h add

`// Arduino GIGA WiFi R1
#elif defined(ARDUINO_ARCH_NRF52840)
  #define CORE_NUM_INTERRUPT	NUM_DIGITAL_PINS`