tttapa/Control-Surface #1030: Does it's support ESP32-S3/C3 via USB port?

DRCRecoveryData, 2024-04-07:
Hi,

Can you add support for ESP32-S3/C3 via USB port?

tttapa, 2024-04-07:
The ESP32-S3 is already supported, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html#espressif-esp32-boards-with-usb-support.

The ESP32-C3 does not have a general-purpose USB interface, so it cannot support MIDI over USB.

DRCRecoveryData, 2024-04-07:
> The ESP32-S3 is already supported, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html#espressif-esp32-boards-with-usb-support.
> 
> The ESP32-C3 does not have a general-purpose USB interface, so it cannot support MIDI over USB.

Can you check this: https://github.com/esp32beans/ESP32USBMIDI 
They say that via otg port because esp32-s3 devkit c have 2 type c port

tttapa, 2024-04-07:
>  They say that via otg port because esp32-s3 devkit c have 2 type c port

(micro B, not type C)

But yes, the same applies to Control Surface's MIDI over USB implementation. The "UART" USB port does not (and cannot) support MIDI over USB.

DRCRecoveryData, 2024-04-10:
> >  They say that via otg port because esp32-s3 devkit c have 2 type c port
> 
> (micro B, not type C)
> 
> But yes, the same applies to Control Surface's MIDI over USB implementation. The "UART" USB port does not (and cannot) support MIDI over USB.

Could you update example midi via otg port?

tttapa, 2024-04-10:
Most of the examples already have it enabled. You simply need to instantiate the `USBMIDI_Interface` (which is done in most examples), and then follow the instructions in https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html#espressif-esp32-boards-with-usb-support (assuming you have the latest version of the arduino-esp32 core installed).

DRCRecoveryData, 2024-04-16:
> Most of the examples already have it enabled. You simply need to instantiate the `USBMIDI_Interface` (which is done in most examples), and then follow the instructions in [...]

Could you convert this code to work with control-surface library:
```#include "MIDIUSB.h"
#include <Adafruit_NeoPixel.h>

#define NEO_PIN 10
#define NUM_PXL 32

// Color palette from the first code
const byte _R[128] = {0, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 45, 93, 142, 223, 190, 28, 61, 93, 190, 125, 12, 28, 45, 158, 61, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 12, 28, 45, 158, 61, 28, 61, 93, 190, 125, 45, 93, 142, 223, 190, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 36, 73, 109, 146, 182, 219, 255};
const byte _G[128] = {0, 0, 0, 0, 125, 0, 12, 28, 45, 158, 61, 28, 61, 93, 190, 125, 45, 93, 142, 223, 190, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 45, 93, 142, 223, 190, 28, 61, 93, 190, 125, 12, 28, 45, 158, 61, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 36, 73, 109, 146, 182, 219, 255};
const byte _B[128] = {0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 0, 0, 0, 125, 0, 12, 28, 45, 158, 61, 28, 61, 93, 190, 125, 45, 93, 142, 223, 190, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 61, 125, 190, 255, 255, 45, 93, 142, 223, 190, 28, 61, 93, 190, 125, 12, 28, 45, 158, 61, 36, 73, 109, 146, 182, 219, 255};

Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUM_PXL, NEO_PIN, NEO_GRB + NEO_KHZ800);

// Define AddressMatrix template
template <size_t ROWS, size_t COLS>
class AddressMatrix {
public:
  uint8_t matrix[ROWS][COLS];

  // Constructor
  AddressMatrix(uint8_t data[ROWS][COLS]) {
    for (size_t i = 0; i < ROWS; i++) {
      for (size_t j = 0; j < COLS; j++) {
        matrix[i][j] = data[i][j];
      }
    }
  }
};

// Define NoteButtonMatrix template
template <size_t ROWS, size_t COLS>
class NoteButtonMatrix {
public:
  uint8_t row_pins[ROWS];
  uint8_t col_pins[COLS];
  AddressMatrix<ROWS, COLS> address_matrix;
  uint8_t channel;

  // Constructor
  NoteButtonMatrix(uint8_t row_pins[ROWS], uint8_t col_pins[COLS], AddressMatrix<ROWS, COLS> address_matrix, uint8_t channel)
      : address_matrix(address_matrix) {
    for (size_t i = 0; i < ROWS; i++) {
      this->row_pins[i] = row_pins[i];
    }
    for (size_t j = 0; j < COLS; j++) {
      this->col_pins[j] = col_pins[j];
    }
    this->channel = channel;
  }

  int read() {
    // Implement your button read logic here
    // Return the button states as an integer
    return 0;
  }
};

// The note numbers corresponding to the buttons in the matrix
const uint8_t button_addresses[4][4] = {
  {1, 2, 3, 4},
  {5, 6, 7, 8},
  {9, 10, 11, 12},
  {13, 14, 15, 16},
};

// Define your NoteButtonMatrix object
uint8_t row_pins[4] = {2, 3, 4, 5}; // row pins
uint8_t col_pins[4] = {6, 7, 8, 9};  // column pins
NoteButtonMatrix<4, 4> buttonmatrix(row_pins, col_pins, AddressMatrix<4, 4>(button_addresses), 1);

void noteOn(byte channel, byte pitch, byte velocity) {
  midiEventPacket_t noteOn = {0x09, 0x90 | channel, pitch, velocity};
  MidiUSB.sendMIDI(noteOn);
}

void noteOff(byte channel, byte pitch, byte velocity) {
  midiEventPacket_t noteOff = {0x08, 0x80 | channel, pitch, velocity};
  MidiUSB.sendMIDI(noteOff);
}

void setPixels(int pinStates){
  for(int i=0; i<16; i++){
    if(bitRead(pinStates, i)){ // Button down
      byte velocity = 127; // Maximum velocity when button is pressed
      noteOn(0, (36+i), velocity); // Send MIDI note on
      pixels.setPixelColor((i*2), pixels.Color(_R[velocity], _G[velocity], _B[velocity])); // Set LED color based on velocity
      pixels.setPixelColor((i*2)+1, pixels.Color(_R[velocity], _G[velocity], _B[velocity]));
    } else { // Button up
      byte velocity = 0; // Velocity 0 when button is released
      noteOff(0, (36+i), velocity); // Send MIDI note off
      pixels.setPixelColor((i*2), pixels.Color(_R[velocity], _G[velocity], _B[velocity])); // Set LED color based on velocity
      pixels.setPixelColor((i*2)+1, pixels.Color(_R[velocity], _G[velocity], _B[velocity]));
    }
  }
  pixels.show();
}

void setup() {
  pixels.begin();
  Serial.begin(115200);
}

void loop(){
  int pinStates = buttonmatrix.read();
  
  // Check for button state changes
  setPixels(pinStates);
}
```

tttapa, 2024-04-16:
Simply replace your own NoteButtonMatrix class by Control Surface's version. Use [this example](https://tttapa.github.io/Control-Surface/Doxygen/d4/d53/NoteButtonMatrix_8ino-example.html) as a starting point.

Then use [NoteButtonMatrix::getPrevState(col, row)](https://tttapa.github.io/Control-Surface/Doxygen/dd/d29/classAH_1_1ButtonMatrix.html#a62dac074b3767f9be91bc05f4e5a0638) to get the button states to turn on/off the appropriate LEDs.