tttapa/Control-Surface #1036: The sketch compiles, but doesn't work

Wallirec, 2024-04-24:
Hello!
Please help me figure it out! I wrote a sketch in the standard MIDI library and it worked (on Arduino Uno with Heduino), I tried to adapt it to your library, the sketch was copied, but the MIDI USB did not work (LСD 1602 displays information correctly). It is very necessary for the sketch to work on Arduino pro micro.

`#include <Control_Surface.h>
USBMIDI_Interface midi;

#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 16, 2);

byte patchNum = 0;

void setup() {
  lcd.begin();
  lcd.backlight();
  lcd.clear();
  lcd.setCursor(0, 0);
  lcd.print("0   ROLAND FP30x");
  delay(500);
  
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  Control_Surface.begin();

}

void loop() {
  if (digitalRead(4) == LOW && patchNum <= 127) {
    patchNum++;
    Control_Surface.sendProgramChange({CHANNEL_4},patchNum);
    lcd.setCursor(0, 0);
    lcd.print("    ROLAND FP30x");
    lcd.setCursor(0, 0);
    lcd.print(String(patchNum));
    delay(200);
  }
  if (digitalRead(5) == LOW && patchNum > 0) {
    patchNum--;
    Control_Surface.sendProgramChange({CHANNEL_4},patchNum);
    lcd.setCursor(0, 0);
    lcd.print("    ROLAND FP30x");
    lcd.setCursor(0, 0);
    lcd.print(String(patchNum));
    delay(200);
  }
  Control_Surface.loop();
}`

tttapa, 2024-04-25:
Please provide more details: what does not work? Does the Arduino not show up as a USB MIDI device? Does it show up without sending messages? Did you use a MIDI monitor to verify? Did you try the `USBDebugMIDI_Interface` with the serial monitor?

Wallirec, 2024-04-25:
Reaper sees the board as an Arduino Micro, but not a MIDI monitor, and Reaper does not accept MIDI commands. Midi monitor (Morningstar) sees the board as "Arduino SA :: Arduino Micro"
  On the screen of the connected LSD screen, the patches change.
I corrected the first message - inserted the sketch as code.

Wallirec, 2024-04-25:
Please forgive me - I'm an idiot!!! The buttons were soldered onto other pins!

closed 2024-04-26 by Wallirec