tttapa/Control-Surface #1040: Error when compiling for ESP32 S3

marcus-universe, 2024-05-10:
**Describe the bug**

I getting this error when compiling my code
```
Compiling sketch...
"C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f312/esp32s3/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" "-DARDUINO_VARIANT=\"esp32s3\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=default,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f312/esp32s3/flags/defines" "-IH:\\Projects\\Coding\\CVRScrips\\ESP32S3\\MyMidicontroller" -iprefix "C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f312/esp32s3/include/" "@C:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f312/esp32s3/flags/includes" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f312/esp32s3/qio_qspi/include" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc1\\cores\\esp32" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc1\\variants\\esp32s3" "-Ic:\\Users\\User\\Documents\\Arduino\\libraries\\Control_Surface\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc1\\libraries\\SPI\\src" "-IC:\\Users\\User\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc1\\libraries\\BLE\\src" "@C:\\Users\\User\\AppData\\Local\\Temp\\arduino\\sketches\\E92C8A90C45645F4F4684F567A2662BD/build_opt.h" "@C:\\Users\\User\\AppData\\Local\\Temp\\arduino\\sketches\\E92C8A90C45645F4F4684F567A2662BD/file_opts" "C:\\Users\\User\\AppData\\Local\\Temp\\arduino\\sketches\\E92C8A90C45645F4F4684F567A2662BD\\sketch\\MyMidicontroller.ino.cpp" -o "C:\\Users\\User\\AppData\\Local\\Temp\\arduino\\sketches\\E92C8A90C45645F4F4684F567A2662BD\\sketch\\MyMidicontroller.ino.cpp.o"
In file included from c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Hardware/Hardware-Types.hpp:8,
                 from c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Hardware/ExtendedInputOutput/ExtendedInputOutput.hpp:14,
                 from c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Hardware/FilteredAnalog.hpp:8,
                 from c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/Control_Surface/Control_Surface_Class.hpp:6,
                 from c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/Control_Surface.h:16,
                 from H:\Projects\Coding\CVRScrips\ESP32S3\MyMidicontroller\MyMidicontroller.ino:1:
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:282:1: error: no declaration matches 'AH::ArraySlice<T, (AH::abs_diff<unsigned int>(Start, End) + 1), (End < Start), false> AH::Array<T, N>::slice()'
  282 | Array<T, N>::slice() {
      | ^~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:138:66: note: candidates are: 'template<class T, unsigned int N> template<unsigned int Start, unsigned int End> AH::ArraySlice<T, (AH::abs_diff<unsigned int>(Start, End) + 1), (End < Start), true> AH::Array<T, N>::slice() const'
  138 |     ArraySlice<T, abs_diff(Start, End) + 1, (End < Start), true> slice() const;
      |                                                                  ^~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:131:67: note:                 'template<class T, unsigned int N> template<unsigned int Start, unsigned int End> AH::ArraySlice<T, (AH::abs_diff<unsigned int>(Start, End) + 1), (End < Start), false> AH::Array<T, N>::slice()'
  131 |     ArraySlice<T, abs_diff(Start, End) + 1, (End < Start), false> slice();
      |                                                                   ^~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:36:8: note: 'struct AH::Array<T, N>' defined here
   36 | struct Array {
      |        ^~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:291:1: error: no declaration matches 'AH::ArraySlice<T, (AH::abs_diff<unsigned int>(Start, End) + 1), (End < Start), true> AH::Array<T, N>::slice() const'
  291 | Array<T, N>::slice() const {
      | ^~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:138:66: note: candidates are: 'template<class T, unsigned int N> template<unsigned int Start, unsigned int End> AH::ArraySlice<T, (AH::abs_diff<unsigned int>(Start, End) + 1), (End < Start), true> AH::Array<T, N>::slice() const'
  138 |     ArraySlice<T, abs_diff(Start, End) + 1, (End < Start), true> slice() const;
      |                                                                  ^~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:131:67: note:                 'template<class T, unsigned int N> template<unsigned int Start, unsigned int End> AH::ArraySlice<T, (AH::abs_diff<unsigned int>(Start, End) + 1), (End < Start), false> AH::Array<T, N>::slice()'
  131 |     ArraySlice<T, abs_diff(Start, End) + 1, (End < Start), false> slice();
      |                                                                   ^~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:36:8: note: 'struct AH::Array<T, N>' defined here
   36 | struct Array {
      |        ^~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:300:1: error: no declaration matches 'AH::ArraySlice<T, (AH::abs_diff<unsigned int>(End, Start) + 1), ((Reverse ^ (End < Start)) != 0), Const> AH::ArraySlice<T, N, Reverse, Const>::slice() const'
  300 | ArraySlice<T, N, Reverse, Const>::slice() const {
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:273:5: note: candidate is: 'template<class T, unsigned int N, bool Reverse, bool Const> template<unsigned int Start, unsigned int End> AH::ArraySlice<T, (AH::abs_diff<unsigned int>(End, Start) + 1), ((Reverse ^ (End < Start)) != 0), Const> AH::ArraySlice<T, N, Reverse, Const>::slice() const'
  273 |     slice() const;
      |     ^~~~~
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/AH/Containers/Array.hpp:168:7: note: 'class AH::ArraySlice<T, N, Reverse, Const>' defined here
  168 | class ArraySlice {
      |       ^~~~~~~~~~
In file included from c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/MIDI_Interfaces/BluetoothMIDI_Interface.hpp:3,
                 from c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/Control_Surface.h:123:
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/MIDI_Interfaces/BLEMIDI.hpp: In member function 'std::string CS::BLEMIDI::getValue()':
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/MIDI_Interfaces/BLEMIDI.hpp:97:62: error: could not convert 'BLECharacteristic::getValue()()' from 'String' to 'std::string' {aka 'std::__cxx11::basic_string<char>'}
   97 |     std::string getValue() { return pCharacteristic->getValue(); }
      |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~^~
      |                                                              |
      |                                                              String
c:\Users\User\Documents\Arduino\libraries\Control_Surface\src/MIDI_Interfaces/BluetoothMIDI_Interface.hpp: In member function 'virtual void CS::BluetoothMIDI_Interface::onWrite(BLECharacteristic*)':
[... cut ...]
- Post your code between triple back ticks with the `cpp` language specifier, as shown below. This ensures that the code is displayed correctly and that syntax highlighting is enabled. (See the [GitHub Markdown guide](https://guides.github.com/features/mastering-markdown/).)

```cpp
#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;

// Instantiate a NoteButton object
CCButton button {5,{MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

tttapa, 2024-05-10:
Which version of Control Surface are you using? I'm unable to reproduce the issue with the latest `main` version.

marcus-universe, 2024-05-12:
> Which version of Control Surface are you using? I'm unable to reproduce the issue with the latest `main` version.

I use [Version 1.2.0](https://github.com/tttapa/Control-Surface/releases/tag/1.2.0)

I also started the Ardinio IDE as an administrator, because if not I can't install Control Surface as a library.

marcus-universe, 2024-05-12:
I tried to install instead the main repo instead of the release. Now its working 🙌

closed 2024-05-12 by marcus-universe