tttapa/Control-Surface #1047: Using Control Surface with Teensy 4.1 board

RandyRyanG, 2024-05-29:
Getting the following errors when I include the control surface library.

Select Teensy 4.1 as board
Try to verify any of the control surface examples.

Should code verify with out issue?

**Code**
- Please post the _full_ code required to reproduce the problem.
- If the code doesn't do what you expect, include a detailed description of what you expected the code it to do.
- Don't post code that is too long. Try to narrow down the problem, try one component or aspect of your project at a time.
- Don't post snippets, always post a complete (compilable) sketch.
- Post your code between triple back ticks with the `cpp` language specifier, as shown below. This ensures that the code is displayed correctly and that syntax highlighting is enabled. (See the [GitHub Markdown guide](https://guides.github.com/features/mastering-markdown/).)

```cpp
// In file included from C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\libraries\SPI/SPI.h:23,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Hardware/LEDs/MAX7219_Base.hpp:8,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Hardware/LEDs/MAX7219SevenSegmentDisplay.hpp:3,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/Control_Surface.h:19,
                 from C:\Users\randy\AppData\Local\Temp\.arduinoIDE-unsaved2024429-16576-2p7owe.sr72\NoteButtonMatrix\NoteButtonMatrix.ino:34:
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/DMAChannel.h: In copy constructor 'DMASetting::DMASetting(const DMASetting&)':
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/DMAChannel.h:399:25: warning: implicitly-declared 'constexpr DMASetting& DMASetting::operator=(const DMASetting&)' is deprecated [-Wdeprecated-copy]
  399 |                 *this = c;
      |                         ^
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/DMAChannel.h:397:9: note: because 'DMASetting' has user-provided 'DMASetting::DMASetting(const DMASetting&)'
  397 |         DMASetting(const DMASetting &c) {
      |         ^~~~~~~~~~
In file included from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/Control_Surface.h:98,
                 from C:\Users\randy\AppData\Local\Temp\.arduinoIDE-unsaved2024429-16576-2p7owe.sr72\NoteButtonMatrix\NoteButtonMatrix.ino:34:
c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/MIDI_Inputs/MCU/VU.hpp: At global scope:
c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/MIDI_Inputs/MCU/VU.hpp:188:7: warning: 'cs::MCU::VU' has a field 'cs::MCU::VU::decayTimer' whose type uses the anonymous namespace [-Wsubobject-linkage]
  188 | class VU : public MatchingMIDIInputElement<MIDIMessageType::ChannelPressure,
      |       ^~
In file included from C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/WProgram.h:58,
                 from C:\Users\randy\AppData\Local\Temp\arduino\sketches\524D84EAEDA2A1D34F17FC939CBC56EE/pch/Arduino.h:6,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Math/FixArduinoMacros.hpp:6,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Math/MinMaxFix.hpp:3,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Arduino-Wrapper.h:3,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src\AH\Math\Quaternion.hpp:24,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src\AH\Math\Quaternion.cpp:1:
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h: In member function 'void usb_midi_class::setHandleSysEx(void (*)(const uint8_t*, uint16_t, bool))':
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h:347:47: warning: cast between incompatible function types from 'void (*)(const uint8_t*, uint16_t, bool)' {aka 'void (*)(const unsigned char*, short unsigned int, bool)'} to 'void (*)(const uint8_t*, uint16_t, uint8_t)' {aka 'void (*)(const unsigned char*, short unsigned int, unsigned char)'} [-Wcast-function-type]
  347 |                 usb_midi_handleSysExPartial = (void (*)(const uint8_t *, uint16_t, uint8_t))fptr;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h: In member function 'void usb_midi_class::setHandleSystemExclusive(void (*)(const uint8_t*, uint16_t, bool))':
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h:351:47: warning: cast between incompatible function types from 'void (*)(const uint8_t*, uint16_t, bool)' {aka 'void (*)(const unsigned char*, short unsigned int, bool)'} to 'void (*)(const uint8_t*, uint16_t, uint8_t)' {aka 'void (*)(const unsigned char*, short unsigned int, unsigned char)'} [-Wcast-function-type]
  351 |                 usb_midi_handleSysExPartial = (void (*)(const uint8_t *, uint16_t, uint8_t))fptr;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/WProgram.h:58,
                 from C:\Users\randy\AppData\Local\Temp\arduino\sketches\524D84EAEDA2A1D34F17FC939CBC56EE/pch/Arduino.h:6,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Math/FixArduinoMacros.hpp:6,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Math/MinMaxFix.hpp:3,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Arduino-Wrapper.h:3,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/PrintStream/PrintStream.hpp:8,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/AH/Debug/Debug.hpp:5,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src\AH\Error\Error.hpp:5,
                 from c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src\AH\Error\Exit.cpp:3:
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h: In member function 'void usb_midi_class::setHandleSysEx(void (*)(const uint8_t*, uint16_t, bool))':
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h:347:47: warning: cast between incompatible function types from 'void (*)(const uint8_t*, uint16_t, bool)' {aka 'void (*)(const unsigned char*, short unsigned int, bool)'} to 'void (*)(const uint8_t*, uint16_t, uint8_t)' {aka 'void (*)(const unsigned char*, short unsigned int, unsigned char)'} [-Wcast-function-type]
  347 |                 usb_midi_handleSysExPartial = (void (*)(const uint8_t *, uint16_t, uint8_t))fptr;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h: In member function 'void usb_midi_class::setHandleSystemExclusive(void (*)(const uint8_t*, uint16_t, bool))':
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h:351:47: warning: cast between incompatible function types from 'void (*)(const uint8_t*, uint16_t, bool)' {aka 'void (*)(const unsigned char*, short unsigned int, bool)'} to 'void (*)(const uint8_t*, uint16_t, uint8_t)' {aka 'void (*)(const unsigned char*, short unsigned int, unsigned char)'} [-Wcast-function-type]
  351 |                 usb_midi_handleSysExPartial = (void (*)(const uint8_t *, uint16_t, uint8_t))fptr;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

tttapa, 2024-05-29:
Hi, these all appear to be warnings, not errors. Does the compilation fail elsewhere?

```cpp
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/DMAChannel.h:399:25: warning: implicitly-declared 'constexpr DMASetting& DMASetting::operator=(const DMASetting&)' is deprecated [-Wdeprecated-copy]
```
```cpp
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h: In member function 'void usb_midi_class::setHandleSysEx(void (*)(const uint8_t*, uint16_t, bool))':
C:\Users\randy\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.59.0\cores\teensy4/usb_midi.h:347:47: warning: cast between incompatible function types from 'void (*)(const uint8_t*, uint16_t, bool)' {aka 'void (*)(const unsigned char*, short unsigned int, bool)'} to 'void (*)(const uint8_t*, uint16_t, uint8_t)' {aka 'void (*)(const unsigned char*, short unsigned int, unsigned char)'} [-Wcast-function-type]
 ```

These are warnings in the Teensy Core, outside of Control Surface. I opened an issue back in 2022, but it has not been addressed: https://github.com/PaulStoffregen/cores/issues/660

```cpp
c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/MIDI_Inputs/MCU/VU.hpp: At global scope:
c:\Users\randy\Documents\Arduino\libraries\Control_Surface\src/MIDI_Inputs/MCU/VU.hpp:188:7: warning: 'cs::MCU::VU' has a field 'cs::MCU::VU::decayTimer' whose type uses the anonymous namespace [-Wsubobject-linkage]
```
This is a valid warning, caused by the fact that `millis` in an inline function on Teensy, but it should not cause any issues.

RandyRyanG, 2024-05-29:
No, is it ok to ignore them? Or turn those warnings off in preferences?

tttapa, 2024-05-29:
Yes, you can ignore them.