tttapa/Control-Surface #1048: BTstackBackgroundBackend connection issue

matthias-glatthorn, 2024-06-03:
**Describe the bug**
I loaded a slightly modified version of [BLEMIDI-Adapter.ino](https://github.com/tttapa/Control-Surface/blob/main/examples/3.%20MIDI%20Interfaces/BLEMIDI-Adapter/BLEMIDI-Adapter.ino) onto my pico w. In [Midi Wrench](https://crudebyte.com/mobile/midi_wrench/) on my iPhone I can see a Bluetooth Midi Device with the name "MIDI Adapter". But when I try to establish a connection, the state goes right back to "Not Connected". When I try the same thing in [Blue See](https://apps.apple.com/de/app/bluesee-ble-debugger/id1336679524?mt=12) I get this log:

```
23:14:33.3070: Device (UUID) state changed to 'Connecting'.

23:14:33.6980: Device (UUID) failed to connect in underlying BLE layer: Error Domain=CBErrorDomain Code=14 "Peer removed pairing information" UserInfo={NSLocalizedDescription=Peer removed pairing information}

23:14:33.6990: Device (UUID) state changed to 'Disconnected' with error: Peer removed pairing Information
```
I am kind of new to C++ and I am missing an example on how to use BTstackBackgroundBackend with GenericBLEMIDI_Interface ...

**To Reproduce**

1. Flash pico w with the provided code
2. Try to connect to the ble midi device ("MIDI Adapter") using [Blue See](https://apps.apple.com/de/app/bluesee-ble-debugger/id1336679524?mt=12) or [Midi Wrench](https://crudebyte.com/mobile/midi_wrench/)

**Expected behavior**
A connection to midi device ("MIDI Adapter") is established

**Code**

```cpp

#include <Control_Surface.h>
#include <MIDI_Interfaces/BLEMIDI/BTstackBackgroundBackend.hpp>
#include <MIDI_Interfaces/GenericBLEMIDI_Interface.hpp>

// Instantiate a MIDI over BLE interface
BTstackBackgroundBackend backend;
GenericBLEMIDI_Interface<BTstackBackgroundBackend> midi_ble;
// Instantiate a 5-pin DIN MIDI interface (on the TX and RX pins of Serial1)
HardwareSerialMIDI_Interface midi_ser {Serial1};
// Instantiate the pipe to connect the two interfaces
BidirectionalMIDI_Pipe pipes;

void setup() {
  // Change the name of the BLE device (must be done before initializing it)
  midi_ble.setName("MIDI Adapter");
  // Manually route MIDI input from the serial interface to the BLE interface,
  // and the MIDI input from the BLE interface to the serial interface
  midi_ser | pipes | midi_ble;
  // Initialize the MIDI interfaces
  MIDI_Interface::beginAll();
  // Initialize the built-in LED
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  // Continuously poll all interfaces and route the traffic between them
  MIDI_Interface::updateAll();
  // Display the connection status using the built-in LED
  digitalWrite(LED_BUILTIN, midi_ble.isConnected() ? HIGH : LOW);
}
```

tttapa, 2024-06-04:
Unfortunately, I don't have any Apple hardware to reproduce this myself.
A quick google search brings up this page: https://forums.developer.apple.com/forums/thread/132488 Does removing the device in the settings resolve the issue?

If the Pico is in fact not storing the necessary pairing information, perhaps it could be related to https://github.com/bluekitchen/btstack/issues/498?

matthias-glatthorn, 2024-06-05:
I just did some testing with my windows 10 computer - I can see a Bluetooth Midi Device with the name "MIDI Adapter" on this computer as well but I get the same behavior as on my iPhone: the computer tries for some time to establish a connection and then goes right back to "Not Connected".

Can anyone confirm weather I got this part right `GenericBLEMIDI_Interface<BTstackBackgroundBackend> midi_ble;` ?

tttapa, 2024-06-06:
Last time I tested it, Windows did not support MIDI over BLE natively, it required a third-party program to be able to connect to MIDI devices, so this could be another issue.

That said, I vaguely remember sometimes having to connect twice on Android after uploading a new sketch. This could be related.

I'm afraid I won't have much time in the near future to investigate this further, but it would be interesting to compare the behavior to the [BTStack examples](https://github.com/raspberrypi/pico-examples/tree/master/pico_w/bt), to rule out any Control Surface-specific issues.

You can find all BLE logic here: https://github.com/tttapa/Control-Surface/blob/main/src/MIDI_Interfaces/BLEMIDI/BTstack/gatt_midi.cpp  
Advertising is configured here: https://github.com/tttapa/Control-Surface/blob/main/src/MIDI_Interfaces/BLEMIDI/BTstack/advertising.cpp  
These are the only files that interact with the low-level BTStack code directly. You can [enable debugging logs](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d99/Debug_8ino-example.html) to see all the events.

> Can anyone confirm weather I got this part right `GenericBLEMIDI_Interface<BTstackBackgroundBackend> midi_ble;` ?

Yes, but why is this necessary? This should be similar to what `BluetoothMIDI_Interface` already does for you:

https://github.com/tttapa/Control-Surface/blob/3aaab29ec58d6e71a20fde08a40531cf32318591/src/MIDI_Interfaces/BluetoothMIDI_Interface.hpp#L77

matthias-glatthorn, 2024-06-06:
> Last time I tested it, Windows did not support MIDI over BLE natively, it required a third-party program to be able to connect to MIDI devices, so this could be another issue.

I tried again with [Bluetooth LE Explorer](https://apps.microsoft.com/detail/9n0ztkf1qd98?hl=de-de&gl=DE) and according to [this video](https://www.youtube.com/watch?v=wDtcvXeYnN0&t=7s), Bluetooth LE Explorer should be capable of establishing MIDI over BLE connections. But I could not establish a connection with Bluetooth LE Explorer either. I got "Failed - Pairing error".

> but it would be interesting to compare the behavior to the [BTStack examples](https://github.com/raspberrypi/pico-examples/tree/master/pico_w/bt), to rule out any Control Surface-specific issues.

I'm not that firm with C++ and building Makefiles ... but I managed to get the `BLECircle.ino` working (Examples > MouseBLE > BLECircle in Arduino IDE)

```
/* Earle F. Philhower, III <earlephilhower@yahoo.com> */
/* Released to the public domain */

#include <MouseBLE.h>

void setup() {
  Serial.begin(115200);
  MouseBLE.begin("CircleBLE Mouse");
  delay(5000);
  Serial.printf("Press BOOTSEL to move the mouse in a circle\n");
}

void loop() {
  if (BOOTSEL) {
    Serial.println("BARREL ROLL!!!");
    float r = 100;
    float ox = 0.0;
    float oy = 0.0;
    for (float a = 0; a < 2.0 * 3.14159; a += 0.1) {
      float ax = r * cos(a);
      float ay = r * sin(a);
      float dx = ax - ox;
      float dy = ay - oy;
      MouseBLE.move(dx, dy, 0);
      ox = ax;
      oy = ay;
      delay(10);
    }
    MouseBLE.setBattery(random(0, 101)); // Set between 0...100%
    while (BOOTSEL) {
      delay(1);
    }
  }
}
```

Thank you for your help so far ... if I manage make progress on this issue, I will report back in this thread :)

tttapa, 2024-06-08:
I borrowed a friend's iPad (8th gen, iPadOS 17.5.1), but I can't seem to be able to reproduce the issue: I've uploaded the code you posted to a Pi Pico W, opened MIDI Wrench on the iPad, and I was able to connect (and disconnect and reconnect, even after power a cycle) without any problem.  
I don't have access to a mac to try BlueSee.

tttapa, 2024-06-08:
Alright, progress, I looked into the problem on Windows, and I could reproduce the issue. Explicitly handling the security manager `JUST_WORKS_REQUEST` event fixes it for me: 9c530d4a8e2aec94126664131326418a128f68ea