tttapa/Control-Surface #1049: Map on Filtered Analog

diegogauffin, 2024-06-08:
Hi!
I am making a CC MIDI pedal and read that to apply mapping function it has to be done by a particular way when the analog has been filtered , right? Could you give me an example ? With the way i am doing it i get a very erratic response. 

#include <Control_Surface.h> // Include the Control Surface library
 
 // Create a filtered analog object on pin A3:
FilteredAnalog<10,      // Output precision in bits
               5,       // The amount of filtering
               uint16_t // The integer type for the calculations
               >
  analog = A3;

// If you want more filtering, you can increase the filter shift factor.
// The number of bits required for the intermediate calculations increases if
// you do so, so you have to use a larger type as well.
FilteredAnalog<10,      // Output precision in bits
               7,       // The amount of filtering
               uint32_t // The integer type for the calculations
               >
  moreFiltering = A3;

// If you don't care about the specific settings, and just want a
// default that works, you can use the following:
FilteredAnalog<> simpleAnalog = A3;

/////////////////////////////////////////////////

// Create a filtered analog object on pin A3:
FilteredAnalog<10,      // Output precision in bits
               5,       // The amount of filtering
               uint16_t // The integer type for the calculations
               >
  analog2 = A6;

// If you want more filtering, you can increase the filter shift factor.
// The number of bits required for the intermediate calculations increases if
// you do so, so you have to use a larger type as well.
FilteredAnalog<10,      // Output precision in bits
               7,       // The amount of filtering
               uint32_t // The integer type for the calculations
               >
  moreFiltering2 = A6;

// If you don't care about the specific settings, and just want a
// default that works, you can use the following:
FilteredAnalog<> simpleAnalog2 = A6;

// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
 
 Bank<5> bank(5);
//   │       └───── number of tracks per bank
//   └───────────── number of banks

// Instantiate a Bank selector to control which one of the four Banks is active.
IncrementSelectorLEDs<5> selector {
  bank, // Bank to manage
  1,  
  {2,5,6,7,15} // push button pin
};

// Instantiate a CCButton object
Bankable::CCButton functionButtons1[] = {
   {{bank, BankType::CHANGE_ADDRESS},  
    0,
   {0x49 ,CHANNEL_1},
   }
  };  

Bankable::CCButtonLatched<5> functionButtons2[] = {
   {{bank, BankType::CHANGE_ADDRESS},  
    19,
   {0x50 ,CHANNEL_1},
   }
  };  

Bankable::CCButtonLatched<5> functionButtons3[] = {
   {{bank, BankType::CHANGE_ADDRESS},  
    20,
   {0x51 ,CHANNEL_1},
   }
  };  

Bankable::CCPotentiometer potentiometer1 {
  {bank, BankType::CHANGE_ADDRESS},      // bank configuration
  A6,                                   // analog pin PEDAL
  {MIDI_CC::General_Purpose_Controller_1 , Channel_1}, // address
};
 
Bankable::CCPotentiometer potentiometer2 {
  {bank, BankType::CHANGE_ADDRESS},      // bank configuration
  A3,                                   // analog pin
  {MIDI_CC::General_Purpose_Controller_2, Channel_1}, // address
}; 

// The maximum value that can be measured (usually 16383 = 2¹⁴-1)
constexpr analog_t maxRawValue = CCPotentiometer::getMaxRawValue();
// The filtered value read when potentiometer is at the 0% position
constexpr analog_t minimumValue = 8700;
// The filtered value read when potentiometer is at the 100% position
constexpr analog_t maximumValue = 13300 ;

 
// A mapping function to eliminate the dead zones of the potentiometer:
// Some potentiometers don't output a perfect zero signal when you move them to
// the zero position, they will still output a value of 1 or 2, and the same
// goes for the maximum position.
analog_t mappingFunction(analog_t raw) {
  // make sure that the analog value is between the minimum and maximum
  raw = constrain(raw, minimumValue, maximumValue);
  // map the value from [minimumValue, maximumValue] to [0, maxRawValue]
  return map(raw, minimumValue, maximumValue, 0, maxRawValue);
}

 
void setup() {

 //   Serial.begin(115200);
 // while (!Serial)
    ;
  // Select the correct ADC resolution
  FilteredAnalog<>::setupADC();
  // If you want, you can add mapping functions to invert the input, for example

  Control_Surface.begin(); // Initialize Control Surface
   potentiometer1.map(mappingFunction);
    potentiometer2.map(mappingFunction);
    pinMode(LED_BUILTIN_TX,INPUT);     //Quita LED Rojo SMD
    pinMode(LED_BUILTIN_RX,INPUT);     //Quita LED Rojo SMD

}
 
void loop() {
 
  Control_Surface.loop(); // Update the Control Surface
   // Read the analog input every millisecond, and print if the value has changed
  //static Timer<millis> timer = 10; // ms
  //if (timer && analog.update())
  //  Serial.println(analog.getValue());
  //  static Timer<millis> timer2 = 10; // ms
  //if (timer2 && analog2.update())
  //  Serial.println(analog2.getValue());

  
  
   analog.update();//reads the analog input, applies the filters,
   analog2.update(); // reads the analog input, applies the filters,

 
   
}

tttapa, 2024-06-08:
Your mapping function is not that extreme, it shouldn't matter much at which point you apply it. Are you sure this is not a hardware or wiring issue? What does the Serial Plotter look like when using the FilteredAnalog example?

diegogauffin, 2024-06-13:
I ve recently solved it ... don´t know why, but in a desk computer when the controller is on the flor i get noisy signal on CC values, but when i pick it , it goes off. I ve  tested on diferents notebooks and works well with less Filtering