tttapa/Control-Surface #1051: Setting CCPotentiometer range

Rolbista, 2024-06-13:
Is it possible to set a range for a potentiometer? Something similar to Behringer BCR2000 where you can set min value and max value when configuring a knob. If it is possible, is it possible to update it dynamically, ie. during a controller's operation, like selecting a bank?

**Describe the solution you'd like**
I want to be able to set a minimal and max CC value sent by a controller (potentiometer).
Let's say a synthesizer has negative modulation from 0 to 63 and positive modulation from 64 to 127. I would like to be able to set my controller to send CC messages only between 0 and 63.

Another example: I want to control a synthesizer filter cutoff, but only in a range betweet 74 and 98 (arbitrary range).

I imagine it as either being able to update a property on a CCPotentiometer class, or entering some kind of config mode to set it.

**Describe alternatives you've considered**
I could try setting a range by hardware on analog level (lowering 5V sent to a potentiometer as reference). That would be imprecise would not allow for arbitrary range setting.

**Additional context**
[I already use banks to change CC number on a pot, here is the code for reference:](https://github.com/Rolbista/MidiBoi/blob/master/switcher_one_pot.ino)

diegogauffin, 2024-06-13:
Did you try the Map function?
 https://github.com/tttapa/Control-Surface/blob/main/examples.old/medium-old/MIDIOutput/Control-Change/CCPotentiometer-Map/CCPotentiometer-Map.ino

tttapa, 2024-06-14:
Using the map function is indeed the recommended approach. See https://github.com/tttapa/Control-Surface/discussions/483#discussioncomment-8574289 for inspiration. (In your case, you would use also change the last two arguments to the Arduino `map` function.)

Rolbista, 2024-06-18:
Hi, thanks for your suggestions, I managed to get it to work, at first statically and then using the template method from thread #483 
Looks like I can update ranges at runtime, but I have to pass the index literally to the template:
```
      potentiometers[whichPot - 1].map(mappingFunction<0>);
```

I still need to figure out how to pass the index dynamically, I wrote this but it doesn't compile:
https://github.com/Rolbista/MidiBoi/pull/9/files#diff-b504ffd5a862888858ea88879a136a63bd6cbe7d35c02d22570cf24d58dd6c55R226

`      minimum_values[whichPot - 1] = 35;
      maximum_values[whichPot - 1] = 100;
      whichPotTemp = whichPot - 1;
      potentiometers[whichPot - 1].map(mappingFunction<whichPotTemp>);

      showLastCC = true;  `
      
Can you suggest an approach here?