tttapa/Control-Surface #1059: Pitch Bend Doesn't Send Zero At Center (Sends +8)

nelsonii, 2024-07-13:
_First off, thank you for this extensive library. I'm using it on a number of MIDI Controllers (mainly PB and CC devices) and it's been extremely helpful. The world needs more Pitch Bend Wheels. :-)_

What I'm seeing is that the last value sent for Pitch Bend -- when it's centered -- is +8 or -9. I see this in the most basic "hello world" PB example and in my more complicated code (using mapping, to correct for joystick deadzones, etc).

My customer is using my controller MainStage on a Mac Pro Powerbook osX 14.5. I see it in Pocket MIDI and MIDI-OX (I use those to test / debug messages).

The "last value of 8" is detuning their DAW. Is there any way to get it to send zero? I use mapping, so I can force a pitch bend send when "centered" if need be -- but I'm not sure what that code would be.

12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	-1354
12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	-1082
12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	-921
12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	-809
12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	-745
12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	-697
12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	-665
**12:34:51.026	From Seeed XIAO M0	Pitch Wheel	1	8**
12:34:52.015	From Seeed XIAO M0	Pitch Wheel	1	-313
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-297
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-265
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-233
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-201
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-153
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-121
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-89
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-57
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	-25
**12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	8**
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	56
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	88
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	120
12:34:52.041	From Seeed XIAO M0	Pitch Wheel	1	168

Here are some of the features I'm using -- just snippits as there is a lot of code. (Again, same behavior using the most basic PB example found in the library.)

*** global ***

```
Bankable::PBPotentiometer potPB {{bankChannel, BankType::ChangeChannel}, pinPB, CHANNEL_1}; // pitch bend

FilteredAnalog<14, 10, uint32_t, uint32_t> filterPB = pinPB;
```

*** setup ***

```
  FilteredAnalog<>::setupADC();
  filterPB.resetToCurrentValue();

    potPB.map(mappingFunctionPB_INV);
```

*** mapping function ***

```
analog_t mappingFunctionPB_INV(analog_t raw) {

    //INVERTING
    raw = constrain(raw, PBminimumValue, PBmaximumValue);
    if (raw<(PBcenter-PBdeadzone) || raw>(PBcenter+PBdeadzone)) { 
      raw = map(raw, PBminimumValue, PBmaximumValue, 16383, 0);
    } else { raw = 8192; }
    return raw;

}
```

tttapa, 2024-07-13:
You could try something like this: (untested)

```cpp
#include <Control_Surface.h>

template <uint8_t InputPrecisionBits>
struct PitchBendSenderMiddle {
    static void send(uint16_t value, MIDIAddress address) {
        value = AH::increaseBitDepthMiddle<14, precision(), uint16_t>(value);
        Control_Surface.sendPitchBend(address.getChannelCable(), value);
    }

    constexpr static uint8_t precision() {
        static_assert(InputPrecisionBits <= 14, "");
        return InputPrecisionBits;
    }
};

struct PBPotentiometerMiddle : MIDIFilteredAnalog<PitchBendSenderMiddle<14>> {
    PBPotentiometerMiddle(pin_t analogPin, MIDIChannelCable address = Channel_1)
        : MIDIFilteredAnalog(analogPin, address, {}) {}
};

USBMIDI_Interface midi;
PBPotentiometerMiddle pot {A0, Channel_1};

void setup() { Control_Surface.begin(); }
void loop() { Control_Surface.loop(); }
```

See this page for details: https://tttapa.github.io/Control-Surface-doc/Doxygen/dd/d5e/group__AH__Math.html#ga274ecbe8f2a095423f25beef281cb9de

[image]

nelsonii, 2024-07-18:
I found that PBPotentiometer getValue() is showing a good "512" (center) 10 bit value. So, I went with a hack like this:

In the main Loop, after Control_Surface.loop();
```
  //Force centering on zero
  if (potPB.getValue()==512 && PBdirty) { 
    Control_Surface.sendPitchBend(Channel(defaultSetting.channelShift) , (uint16_t) 8192); 
    PBdirty=false;
  }
```

(Note: defaultSetting.channelShift is a variable I use to keep track of the current MIDI Channel, as I use banks and allow the user to change channels at will.)

Where PBdirty is a boolean. I set it to "true" in my PB mapping code:

```
analog_t map_PB_INV(analog_t raw) {

    //INVERTING
    raw = constrain(raw, PBminimumValue, PBmaximumValue);
    if (raw<(PBcenter-PBdeadzone) || raw>(PBcenter+PBdeadzone)) { 
      raw = map(raw, PBminimumValue, PBmaximumValue, 16383, 0);
      **PBdirty = true;**
    } else { raw = 8192; }
    return raw;
}
```

It's not elegant, but now I get a "zero" on pitch bend when centered. (Centered, according to my map, as my physical design does not allow a full "0 to max" throw -- it's a much smaller range. I do a calibration on the device to get the low/high range values that map uses. )

tttapa, 2024-07-18:
Glad to hear that you got it to work.
Did the approach above not work for you?

nelsonii, 2024-07-21:
I was rushing to get a fix for a user, and didn't see a clear way to make it Bankable (I use banks to allow the user to change MIDI Channels). More of a "me not understanding it well enough" than an issue with the code. It's still on my list of things to work on. :-) 

I'm also using the work-around for an issue one user was seeing with his DAW -- it wanted (not sure why) PB to keep sending the min or max values when at the limits. (Not just send "8192" once.) Like continuous send. This could very well be a DAW config issue -- but they are hard to debug with so many DAWs out there.

josehdx, 2026-01-15:
Hello

I would like to get some assistance as i am facing similar issue with my potentiometer

previously was reported to the center or dead some of my potentiometer was not completely centered so i had to add some extra correction using code, but right now as mentioned on this original post, my center is around +/- 8 and this can cause detune using the pith function on DAW

my current code is this:

`#include <Arduino.h>

#include <Control_Surface.h>

#include <AH/Hardware/MultiPurposeButton.hpp>

BluetoothMIDI_Interface midi;
//USBMIDI_Interface midi;

// Instantiate a PBPotentiometer object
PBPotentiometer potentiometer {
  15,        // Analog pin connected to potentiometer
  Channel_1, // MIDI Channel 1
};

// Mapping Pitchbend
constexpr analog_t minimumValue = 0;
constexpr analog_t maximumValue = 16383; 
constexpr analog_t margin = 375;

analog_t mappingFunction(analog_t raw) {
  raw = constrain(raw, minimumValue, maximumValue);
  if (raw <= 8192 - margin)
    return map(raw, minimumValue, 8192 - margin, 0, 8192 + 64); 
  else if (raw >= 8192 + margin)
    return map(raw, 8192 + margin, maximumValue, 8192 + 64, 16383);
  else
    return 8192 + 64;
}

void setup() {

  Control_Surface.begin();
  potentiometer.map(mappingFunction); // Add the mapping function to the potentiometer

}

void loop() {

  Control_Surface.loop();

  delay(5); 
  
}
`

I tried both proposed solutions being the one from @nelsonii  a little harder to understand as i am not using banks

The one from @tttapa seems more simple but it is not working as it is, maybe i am implemeting something wrong, and actually i am not eve sure what is this thing "PBPotentiometerMiddle"

I currently have "PBPotentiometer" as an objet but not sure how to implement in the right way

Any assistance is highly appreciated

Regards

nelsonii, 2026-01-15:
1) In my PB mapping function, in addition to the mapping result, I set a variable I call "PBwasOffCenter" to true. That helps me keep track if the PB was shifted outside of the deadzone (i.e. your [margin] variable).

In the below map, PBdeadzone is like your margin. I have the LowerShift and UpperShift if I have to tweak things, but currently those are set to zero.

```
`//=================================================================================

analog_t map_PB(analog_t raw) {

    analog_t result = 0;

    //NON INVERTING
    raw = constrain(raw, PBminimumValue, PBmaximumValue);
    
    if (raw <= PBcenter-PBdeadzone-PBdeadzoneLowerShift) {
      result = map(raw, PBminimumValue, PBcenter-PBdeadzone-PBdeadzoneLowerShift, 0, 8191);
      PBwasOffCenter = true;
    }
    else if (raw >= PBcenter+PBdeadzone+PBdeadzoneUpperShift) {
      result = map(raw, PBcenter+PBdeadzone+PBdeadzoneUpperShift, PBmaximumValue, 8191, 16383);
      PBwasOffCenter = true;
    }
    else {
      result = 8192;
    }

    return result;
}

//=================================================================================`
```

2) In the main loop, I call a method I created called adjustPB(). This gets the PB raw value. If it's centered (8192) and the PBwasOffCenter is true (previously off-center), then it forces a center to be sent.

Note that this method also does two other things, which you might not need. It does continuous send on low and continuous send on high. You can skip that if you don't need it.

Also, I have a FORCE_CENTER_UPDATE_DELAY constant and delay in there. It's set to a small value (250ms) and is in there to prevent an excess of zeros being sent out. 

In pseudocode:

* get the raw PB value
* is it in center? (8192)
* was it previously off center? (outside deadzone)
* have we waited at least 0.25 seconds? (to prevent overrun)
* if all above true, then sendPitchBend center (8192)

```
`//=================================================================================

void adjustPB() {

  uint32_t pbGetValue = potPB.getValue();
  uint32_t pbGetRawValue = potPB.getRawValue();

  analog_t pbMapRawValue = map_PB(pbGetRawValue);

  //Continuous send on low
  if (pbGetValue==0) { Control_Surface.sendPitchBend(Channel(defaultSetting.channelShift) , (uint16_t) 0); }

  //If it was off center, but now back to center, force a zero (center)
  if (pbMapRawValue==8192 && PBwasOffCenter) {
    //Serial.print("RE-CENTER REQUEST...");
    //Throttle this behavior. Say, X times in last Y seconds? (like flash update delay).
    if ( (millis()-PBlastCenteredOn) > (FORCE_CENTER_UPDATE_DELAY) ) { 
      Control_Surface.sendPitchBend(Channel(defaultSetting.channelShift) , (uint16_t) 8192);
      PBwasOffCenter = false;
      PBlastCenteredOn = millis();
      Serial.println("[FORCE MIDI CENTER]");
    }
  }

  //continuous send on high
  if (pbGetValue==1023) { Control_Surface.sendPitchBend(Channel(defaultSetting.channelShift) , (uint16_t) 16383); }

}//adjustPB

//=================================================================================
`
```

nelsonii, 2026-01-16:
I ripped apart some old code to simplify it -- hardcoded min/max values, etc. The newer code handles more situations, but this gives you an idea.

```
#include <Control_Surface.h> // Using version 2.0.0

//=================================================================================

USBMIDI_Interface usbmidi;

Bank<16> bankChannel;

int pinPB = A3;

Bankable::PBPotentiometer potPB {{bankChannel, BankType::ChangeChannel}, pinPB, Channel_1}; // pitch bend
FilteredAnalog<10, 10, uint32_t, uint32_t> filterPB = pinPB;

//=================================================================================

bool PBdirty = false;
analog_t PBcenteredSample = 8000; // Runtime Sampled Value default
analog_t PBcenter = 8000; // Flash / Calibrated Value
analog_t PBdeadzone = 50;
analog_t PBminimumValue = 6000;
analog_t PBmaximumValue = 10000;

//=================================================================================

analog_t map_PB_NOI(analog_t raw) {

    //NON INVERTING
    raw = constrain(raw, PBminimumValue, PBmaximumValue);
    if (raw<(PBcenter-PBdeadzone) || raw>(PBcenter+PBdeadzone)) { 
      raw = map(raw, PBminimumValue, PBmaximumValue, 0, 16383);
      PBdirty = true;
    } else { raw = 8192; }
    return raw;
}

//=================================================================================

analog_t map_PB_INV(analog_t raw) {

    //INVERTING
    raw = constrain(raw, PBminimumValue, PBmaximumValue);
    if (raw<(PBcenter-PBdeadzone) || raw>(PBcenter+PBdeadzone)) { 
      raw = map(raw, PBminimumValue, PBmaximumValue, 16383, 0);
      PBdirty = true;
    } else { raw = 8192; }
    return raw;
}

//=================================================================================

void setup() {

  //Determine center point
  pinMode(pinPB, INPUT);
  int iNumberOfSamples = 100; long lSampleSum = 0;
  for (int iSample = 1; iSample<=iNumberOfSamples; iSample++) {
    lSampleSum += analogRead(pinPB); delay(5);
  }
  PBcenteredSample=map((int(lSampleSum/iNumberOfSamples)), 0, 1023, 0, 16383);
  PBcenter = PBcenteredSample;
  
  FilteredAnalog<>::setupADC();
  filterPB.resetToCurrentValue();

  // Add the mapping function to the potentiometers
  //IMPORTANT: Do NOT use the INVERT method -- it overrides the map!!!
  potPB.map(map_PB_INV); // pitch bend - inverted

  Control_Surface.begin();

  //Set the default/startup MIDI Channel -- zero based (0 = midi channel 1)
  bankChannel.select(0);

}//setup

//=================================================================================

void loop() {

  Control_Surface.loop();

  //Force centering on zero
  if (potPB.getValue()==512 && PBdirty) { 
    Control_Surface.sendPitchBend(Channel(0) , (uint16_t) 8192); 
    PBdirty=false;
  }

  yield(); delay(1); // helpful for chips with watchdog timers, short pause to allow other tasks to catch up
  
}//loop

//=================================================================================

```