tttapa/Control-Surface #1086: pi pico receive on cable_2

benzman, 2024-10-23:
Hi Pieter,

I cannot receive data on a virtual cable_2. I can send on all virtual cables but only receive on cable_1.

I have a pi pico with MIDI over USB (with Adafruit TinyUSB-Stack). It works fine. Now I want to use a second virtual port on the device. So I found this to activate the second cable:
midi.backend.backend.setCables(2);

I can send Values over the cable_2. For example:
midi.sendNoteOn({MCU::PLAY, Cable_2}, 0x7F);

But I don't get anything back. I tried with NoteValue:
NoteValue playObj{ {MCU::PLAY, Cable_2} };

To reproduce it I tried the examples sketch: 9. Note-FastLED.ino. Again: Works with cable_1 and doesn't with cable_2

```cpp
#include <FastLED.h>
// Must be before Control Surface to enable FastLED features of Control Surface
#include <Control_Surface.h>

// Define the array of leds.
Array<CRGB, 8> leds {};
// The data pin with the strip connected.
constexpr uint8_t ledpin = 22;

USBMIDI_Interface midi;

// Create a MIDI input element that listens to all notes in the range C4 - G4
// (the range starts at C4 and has a length equal to `leds.length` == 8).
NoteRangeFastLED<leds.length> midiled {leds, {MIDI_Notes::C[4], Channel_1, Cable_2}};

void setup() {
  midi.backend.backend.setCables(2);
  // See FastLED examples and documentation for more information.
  FastLED.addLeds<NEOPIXEL, ledpin>(leds.data, leds.length);
  FastLED.setCorrection(TypicalPixelString);
  midiled.setBrightness(128);
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
  if (midiled.getDirty()) { // If the colors changed
    FastLED.show();         // Update the LEDs with the new colors
    midiled.clearDirty();   // Clear the dirty flag
  }
}

```

Can you help me?

tttapa, 2024-10-27:
Multiple cable support for the RP2040 has been enabled in 5f492cf192a24af748d9c9c51e2b5b32c2355396.

benzman, 2024-10-27:
Thanks a lot! That works for my use-case!
Just for fun I tried to define more than three cables on the pi pico and when I flash it, it doesn't show up as a midi device on my Mac. (Flashing works without a problem and everything is fine up to three cables)

tttapa, 2024-10-27:
I believe that this is an issue with the Adafruit TinyUSB library. Their example also doesn't work beyond 3 cables for me: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/3.3.3/examples/MIDI/midi_multi_ports/midi_multi_ports.ino

The Adafruit TinyUSB library appears to have multiple issues that unfortunately do not seem to be addressed by the maintainers. See the links on the following page: https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html#mbed-os-boards-with-native-usb-support.  
I have plans to add support for multiple cables when using the Pico SDK stack, but I have too much other things to do at the moment.

benzman, 2024-10-27:
Okay than. Thanks again for all your work. It is a lot of fun using your library!

closed 2024-10-27 by benzman