tttapa/Control-Surface #1093: Switching between CCButton and CCLatched

diegogauffin, 2024-11-14:
Hey Pieter! How are you? Hope fine! 
I was thinking it would be really usefulll to get a way to change between CCButton (momentary) and CC Latched in the run. Do you think is that posible ? I mean doing it without changing the code, upload and so on.

Wallirec, 2024-12-02:
I don't know how to get a new function without changing the code.
But here is an example of code with a button (connected to pin 3) which, when pressed briefly, transmits CC 20 Latched, and when pressed long, CC20 momentary
```
#include <Control_Surface.h>

USBMIDI_Interface midi;

// Ignore the name “increment” button, it's simply a button
// class that can detect long and short presses.
IncrementButton btn { 3 }; // pin number of button

void setup() {
  Control_Surface.begin();
  btn.begin();
}

void loop() {
  Control_Surface.loop();
  switch (btn.update()) {
    case IncrementButton::ReleasedShort: // If button is released after short press
    Control_Surface.sendCC(24, btn2_flg * 127);
    case IncrementButton::IncrementLong: // If button is still pressed after some time
    Control_Surface.sendCC(24, 127); break;
  }
}
```

gwilherm, 2024-12-02:
You can use pointer to MIDIOutputElement and delete/create instance of CCButton or CCButtonLatched.
If you need to use `getAddress()` or `setAddressUnsafe()` there is no common interface for this but Pieter came with a workaround that helped me a lot: https://github.com/tttapa/Control-Surface/pull/971#issuecomment-1793139526