tttapa/Control-Surface #1094: virtual void cs::MIDIStaller::handleStall()' was hidden [-Werror=overloaded-virtual=]

matthias-glatthorn, 2024-11-16:
**Describe the bug**
After I updated my boards in arduino IDE I started to get this error:

```
WARNING: library Control Surface claims to run on avr, sam, samd, teensy, esp32, esp8266, megaavr, mbed, mbed_nano, mbed_rp2040, renesas_uno architecture(s) and may be incompatible with your current board which runs on rp2040 architecture(s).
In file included from c:\Users\SpecialGuest\Documents\Arduino\libraries\Control_Surface\src/MIDI_Interfaces/MIDI_Interface.hpp:5,
                 from c:\Users\SpecialGuest\Documents\Arduino\libraries\Control_Surface\src/Control_Surface/Control_Surface_Class.hpp:10,
                 from c:\Users\SpecialGuest\Documents\Arduino\libraries\Control_Surface\src/Control_Surface.h:16,
                 from C:\Users\SpecialGuest\AppData\Local\Temp\.arduinoIDE-unsaved20241016-19528-1nkcrcg.18kp\1.First-Output\1.First-Output.ino:12:
c:\Users\SpecialGuest\Documents\Arduino\libraries\Control_Surface\src/MIDI_Interfaces/MIDI_Staller.hpp:18:18: error: 'virtual void cs::MIDIStaller::handleStall()' was hidden [-Werror=overloaded-virtual=]
   18 |     virtual void handleStall() = 0;
      |                  ^~~~~~~~~~~
c:\Users\SpecialGuest\Documents\Arduino\libraries\Control_Surface\src/MIDI_Interfaces/MIDI_Interface.hpp:123:17: note:   by 'cs::MIDI_Interface::handleStall(MIDIInterface_t*)'
  123 |     static void handleStall(MIDIInterface_t *self);
      |                 ^~~~~~~~~~~
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: exit status 1
```
I'm using the latest build of Control Surface. I get this error when I import the Control Surface library `#include <Control_Surface.h>`

**To Reproduce**
Open the `1. First-output` from the examples

**Expected behavior**
Sketch should compile. The above error should not occur.

**Code**

Open the `1. First-output` from the examples with Board `Rasberry Pi Pico W`

matthias-glatthorn, 2024-11-16:
I don't get this Error with 4.0.3 of https://github.com/earlephilhower/arduino-pico/releases

tttapa, 2024-11-16:
I can't seem to be able to reproduce the error on Linux with version 4.2.1 of https://github.com/earlephilhower/arduino-pico and the latest `main` version of Control Surface unless I explicitly add the `-Werror=overloaded-virtual` flag.

Either way, it's a harmless warning, I've just pushed a fix.

commit 317e36a6 (2024-11-16): Don't hide MIDIStaller::handleStall, fixes #1094
closed 2024-11-16 by tttapa