tttapa/Control-Surface #1095: Arduino Lint Error

gwilherm, 2024-11-17:
Hello,
This is not really a feature request but it is also not a bug report,
I would love to see Control-Surface available in the official arduino library registry.
This allows to include the library directly with the arduino IDE, and in projects using [sketch project file](https://arduino.github.io/arduino-cli/1.1/sketch-project-file) that make projects more distributable

For this, there are some rules and the library must pass the linter with no error.
https://github.com/arduino/library-registry/blob/main/FAQ.md#submission-requirements

> Arduino has created a command line tool to check libraries for compliance with all the Library Manager requirements:
> https://github.com/arduino/arduino-lint

The tool detects one error on the library (see the logs below)

I admit that the error is questionable (https://github.com/arduino/arduino-cli/issues/990, https://github.com/arduino/arduino-lint/issues/588) for the `test` directory.
[The solution proposed](https://github.com/arduino/arduino-lint/issues/588#issuecomment-1700684019) is to move the `test` directory into a folder named `extras`.
I understand this is kinda intrusive to change the repository architecture like this, but this would be really nice to find your library in the registry.

```
WARNING: library.properties name Control Surface contains spaces. Although supported, best practices is to not use      
         spaces.                                                                                                        
         See: https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format         
         (Rule LP015)                                                                                                   
ERROR: Sketch(es) found outside examples and extras folders:                                               
         Control-Surface/examples.old/medium-old/Audio/OLED-Demo-Audio                                     
         Control-Surface/examples.old/medium-old/Audio/VUMeterLEDs-USB-DAC                                 
         Control-Surface/examples.old/medium-old/Audio/VUMeterOLED-USB-DAC                                 
         Control-Surface/examples.old/medium-old/Display/OLED-Demo                                         
         Control-Surface/examples.old/medium-old/Display/VU-Bridge                                         
         Control-Surface/examples.old/medium-old/Display/VU-Bridge-Dual-Display                            
         Control-Surface/examples.old/medium-old/ESP32/VU-Bridge-ESP32                                     
         Control-Surface/examples.old/medium-old/ExtendedInputOutput/Ex.1.Shift-Register-LED-Effects       
         Control-Surface/examples.old/medium-old/ExtendedInputOutput/Multiplexer-AnalogReadSerial          
         Control-Surface/examples.old/medium-old/ExtendedInputOutput/PushButtonsToggleLEDsShiftRegMultiplex
         Control-Surface/examples.old/medium-old/ExtendedInputOutput/SPIShiftRegisterBlink                 
         Control-Surface/examples.old/medium-old/ExtendedInputOutput/ShiftRegisterBlink                    
         Control-Surface/examples.old/medium-old/Getting-Started                                           
         Control-Surface/examples.old/medium-old/MIDIInput/LED                                             
         Control-Surface/examples.old/medium-old/MIDIInput/MCU_VULEDs-ShiftRegister                        
         Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCIncrementDecrementButtons     
         Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCPotentiometer                 
         Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCPotentiometer-Map             
         Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCRotaryEncoder                 
         Control-Surface/examples.old/medium-old/MIDIOutput/Enable-disable                                 
         Control-Surface/examples.old/medium-old/MIDIOutput/Notes/NoteButtonMatrixKeyboard                 
         Control-Surface/examples.old/medium-old/MIDIOutput/Notes/NoteChordButton                          
         Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtons                              
         Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtonsLEDs                          
         Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtonsLEDsNoMIDIFeedback            
         Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtonsLEDsShiftRegMultiplex         
         Control-Surface/examples.old/medium-old/MIDI_Interfaces/Bluetooth-BLE-MIDI-CCPotentiometer        
         Control-Surface/examples.old/medium-old/MIDI_Interfaces/Bluetooth-BLE-MIDI-NoteButton             
         Control-Surface/examples.old/medium-old/MIDI_Interfaces/Bluetooth-BLE-MIDI-NoteButton_Input       
         Control-Surface/examples.old/medium-old/Other/MIDI_Callback                                       
         Control-Surface/examples.old/medium-old/ProgramChange/ProgramChanger                              
         Control-Surface/examples.old/medium-old/Transpose/EncoderSelector-Transpose                       
         Control-Surface/examples.old/medium-old/Transpose/EncoderSelectorLEDs-Transpose                   
         Control-Surface/examples.old/medium-old/Transpose/EncoderSelectorLEDs-Transpose-Generate-Array    
         Control-Surface/examples.old/medium-old/Transpose/Transpose                                       
         Control-Surface/test/Construction/ErrorCheckAllMIDIOutputs                                        
         Control-Surface/test/tools/sketches/midi-echo                                                     
         Control-Surface/test/tools/sketches/midi-receive                                                  
         Control-Surface/test/tools/sketches/midi-send                                                     
       See: https://arduino.github.io/arduino-cli/latest/library-specification/#library-examples           
       (Rule LD003)                                                                                        

Linter results for project: 1 ERRORS, 1 WARNINGS
```

Regards,
Gwilherm

per1234, 2025-04-25:
>there are some rules and the library must pass the linter with no error.

This is correct. However, note that it is only required to pass with **Arduino Lint**'s compliance level set to "permissive":

https://arduino.github.io/arduino-lint/latest/commands/arduino-lint/#options

At the "permissive" level, rule violations are only treated as an error if they result in an effective defect in usability.

The rule violation that caused the rejection of the Library Manager submission was:

https://github.com/arduino/library-registry/pull/5444#issuecomment-2481281660

```text
ERROR: Symlink(s) found. Symlinks cause difficulties for Windows users. These block addition to the Arduino Library     
       Manager index:                                                                                                   
         Control-Surface/doxygen/pages/xbm.md                                                                           
       (Rule LS005)                                                                                                     
```

Note that the violation of Arduino Lint rule [LD003](https://arduino.github.io/arduino-lint/latest/rules/library/#stray-sketch-ld003) that you report here was only treated as a warning:

```text
WARNING: Sketch(es) found outside examples and extras folders:                                               
           Control-Surface/examples.old/medium-old/Audio/OLED-Demo-Audio                                     
           Control-Surface/examples.old/medium-old/Audio/VUMeterLEDs-USB-DAC                                 
           Control-Surface/examples.old/medium-old/Audio/VUMeterOLED-USB-DAC                                 
           Control-Surface/examples.old/medium-old/Display/OLED-Demo                                         
           Control-Surface/examples.old/medium-old/Display/VU-Bridge                                         
           Control-Surface/examples.old/medium-old/Display/VU-Bridge-Dual-Display                            
           Control-Surface/examples.old/medium-old/ESP32/VU-Bridge-ESP32                                     
           Control-Surface/examples.old/medium-old/ExtendedInputOutput/Ex.1.Shift-Register-LED-Effects       
           Control-Surface/examples.old/medium-old/ExtendedInputOutput/Multiplexer-AnalogReadSerial          
           Control-Surface/examples.old/medium-old/ExtendedInputOutput/PushButtonsToggleLEDsShiftRegMultiplex
           Control-Surface/examples.old/medium-old/ExtendedInputOutput/SPIShiftRegisterBlink                 
           Control-Surface/examples.old/medium-old/ExtendedInputOutput/ShiftRegisterBlink                    
           Control-Surface/examples.old/medium-old/Getting-Started                                           
           Control-Surface/examples.old/medium-old/MIDIInput/LED                                             
           Control-Surface/examples.old/medium-old/MIDIInput/MCU_VULEDs-ShiftRegister                        
           Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCIncrementDecrementButtons     
           Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCPotentiometer                 
           Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCPotentiometer-Map             
           Control-Surface/examples.old/medium-old/MIDIOutput/Control-Change/CCRotaryEncoder                 
           Control-Surface/examples.old/medium-old/MIDIOutput/Enable-disable                                 
           Control-Surface/examples.old/medium-old/MIDIOutput/Notes/NoteButtonMatrixKeyboard                 
           Control-Surface/examples.old/medium-old/MIDIOutput/Notes/NoteChordButton                          
           Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtons                              
           Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtonsLEDs                          
           Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtonsLEDsNoMIDIFeedback            
           Control-Surface/examples.old/medium-old/MIDIOutput/Notes/PushButtonsLEDsShiftRegMultiplex         
           Control-Surface/examples.old/medium-old/MIDI_Interfaces/Bluetooth-BLE-MIDI-CCPotentiometer        
           Control-Surface/examples.old/medium-old/MIDI_Interfaces/Bluetooth-BLE-MIDI-NoteButton             
           Control-Surface/examples.old/medium-old/MIDI_Interfaces/Bluetooth-BLE-MIDI-NoteButton_Input       
           Control-Surface/examples.old/medium-old/Other/MIDI_Callback                                       
           Control-Surface/examples.old/medium-old/ProgramChange/ProgramChanger                              
           Control-Surface/examples.old/medium-old/Transpose/EncoderSelector-Transpose                       
           Control-Surface/examples.old/medium-old/Transpose/EncoderSelectorLEDs-Transpose                   
           Control-Surface/examples.old/medium-old/Transpose/EncoderSelectorLEDs-Transpose-Generate-Array    
           Control-Surface/examples.old/medium-old/Transpose/Transpose                                       
           Control-Surface/examples.old/very-old/Ex.01.Potentiometer                                         
           Control-Surface/examples.old/very-old/Ex.02.Button                                                
[... cut ...]
The symlink has already been removed from the library: 
https://github.com/tttapa/Control-Surface/commit/ed57f9b22f9e656a9cb7363c812ea9f56e92d5ae. However, Library Manager works exclusively from Git tags, and there hasn't been a tag since the time of the fix. So the checks would continue to fail until a tag has been pushed to the repository that contains a fix. This means that the only thing that needs to be done to allow the addition of the library to Library Manager is for a maintainer to push a tag.

gwilherm, 2025-05-06:
Thank you ! Control-Surface library has now been added to the registry !
https://github.com/arduino/library-registry/pull/6273