tttapa/Control-Surface #1099: Pedal Pitch Bend +- oktava

Wallirec, 2024-11-24:
Thanks for the best library!
I want to make a pitch bend pedal (like Wammy).
What I've got so far: I found (here) and adjusted the sketch for my homemade pedal on a Hall sensor.
`#include <Control_Surface.h>

USBMIDI_Interface midi;

PBPotentiometer potentiometers[] = {
  { A0, CHANNEL_1, },
};

template <analog_t MinVal, analog_t MaxVal>
analog_t mappingFunction(analog_t raw) {
  raw = constrain(raw, MinVal, MaxVal);
  return map(raw, MinVal, MaxVal, 0, FilteredAnalog<>::getMaxRawValue());
}

void setup() {
  potentiometers[0].map(mappingFunction<7900, 14900>);
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}`

Now the pedal lowers the sound by one tone.
I want the sound to be lowered by an octave.
And I would also like (with an additional button) to invert the pedal to increase the pitch.
Please tell me how to do this!

Wallirec, 2024-11-24:
By moving the midpoint from 0 to 7900 and adjusting MaxVal, I adjusted the counter to increase the tone.

```
#include <Control_Surface.h>

USBMIDI_Interface midi;

PBPotentiometer potentiometers[] = {
  { A0, CHANNEL_1, },
};

template <analog_t MinVal, analog_t MaxVal>
analog_t mappingFunction(analog_t raw) {
  raw = constrain(raw, MinVal, MaxVal);
  return map(raw, MinVal, MaxVal, 7900, FilteredAnalog<>::getMaxRawValue());
}

void setup() {
  potentiometers[0].map(mappingFunction<7900, 11900>);
  Control_Surface.begin();
}

void loop() {  
  Control_Surface.loop();
}
```
All that remains is to combine them and expand the pitch range - please advise!

Wallirec, 2024-11-25:
I think I'm starting to guess that the pitch bend range depends on the instrument settings.
I found a midi implementation for my instrument - tell me what I need to add to the sketch to increase the range to +- 24 semitones
[image]

tttapa, 2024-11-25:
(Untested)
```cpp
  Control_Surface.sendControlChange(MIDI_CC::RPN_MSB, 0x00);
  Control_Surface.sendControlChange(MIDI_CC::RPN_LSB, 0x00);
  Control_Surface.sendControlChange(MIDI_CC::Data_Entry_MSB, 12);
  Control_Surface.sendControlChange(MIDI_CC::Data_Entry_LSB, 0);
  Control_Surface.sendControlChange(MIDI_CC::RPN_MSB, 0x7F);
  Control_Surface.sendControlChange(MIDI_CC::RPN_LSB, 0x7F);
```

Wallirec, 2024-11-26:
Pieter, thanks for your response (this is not your library's problem).
Sketch does not compile - error: 'Data_Entry_LSB' is not a member of 'CS::MIDI_CC'
Control_Surface.sendControlChange(MIDI_CC::Data_Entry_LSB, 0);
suggested alternative: 'Data_Entry_MSB'

```
#include <Control_Surface.h>

USBMIDI_Interface midi;

PBPotentiometer potentiometers[] = {
  { A0, CHANNEL_1, },
};

template <analog_t MinVal, analog_t MaxVal>
analog_t mappingFunction(analog_t raw) {
  raw = constrain(raw, MinVal, MaxVal);
  return map(raw, MinVal, MaxVal, 7900, FilteredAnalog<>::getMaxRawValue());
}

void setup() {
  potentiometers[0].map(mappingFunction<7900, 11900>);
  Control_Surface.sendControlChange(MIDI_CC::RPN_MSB, 0x00);
  Control_Surface.sendControlChange(MIDI_CC::RPN_LSB, 0x00);
  Control_Surface.sendControlChange(MIDI_CC::Data_Entry_MSB, 12);
  Control_Surface.sendControlChange(MIDI_CC::Data_Entry_LSB, 0);
  Control_Surface.sendControlChange(MIDI_CC::RPN_MSB, 0x7F);
  Control_Surface.sendControlChange(MIDI_CC::RPN_LSB, 0x7F);
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

tttapa, 2024-11-26:
Are you using the latest version of the library?

Wallirec, 2024-11-26:
I updated the library and all my sketches stopped compiling (they are all on your library).

closed 2024-12-21 by Wallirec