tttapa/Control-Surface #1108: Midi-USB

ashojon, 2025-01-05:
**Describe the bug**
I have an ESP32-S3 board and am trying to build a USB and BLE MIDI controller using the Control Surface library. Both USB and BLE MIDI work perfectly in the Arduino IDE. However, when I use PlatformIO with Visual Studio Code, the USB MIDI does not appear in Windows MIDI monitor software.

That is what I used as setting in Arduino IDE
--------------------------------------------------
Set board to ESP32-S3 in `Tools -> Board -> esp32 -> ESP32S3 Dev Module`
Set USB mode to USB-OTG `Tools -> USB Mode -> USB-OTG (TinyUSB)`
Connect the dev board to the PC (UART USB port)
Select the according port in `Tools -> Port`

That is what I used as Setting in Platformio
-----------------------------------------------
[env:esp32s3usbotg]
platform = espressif32
board = esp32s3usbotg
framework = arduino
board_build.mcu = esp32s3
lib_deps = https://github.com/tttapa/Control-Surface.git#main
lib_ignore = MIDIUSB

**Expected behavior**
Esp32-S3 should recognized as MIDI-USB 

**Code**
- Please post the _full_ code required to reproduce the problem.
- If the code doesn't do what you expect, include a detailed description of what you expected the code it to do.
- Don't post code that is too long. Try to narrow down the problem, try one component or aspect of your project at a time.
- Don't post snippets, always post a complete (compilable) sketch.
- Post your code between triple back ticks with the `cpp` language specifier, as shown below. This ensures that the code is displayed correctly and that syntax highlighting is enabled. (See the [GitHub Markdown guide](https://guides.github.com/features/mastering-markdown/).)

```cpp
#include <Arduino.h>
#include <Control_Surface.h>

USBMIDI_Interface midi;
BluetoothMIDI_Interface midi_ble;

void setup() {
    Serial.begin(460800);
    midi_ble.setName("MIDI_BLE_32");
   
  midi_ble.begin();
  midi.begin(); // Initialize the MIDI interface
}

void loop() {

    // Send a MIDI Note On message for the given note to trigger it
    midi.sendNoteOn(note, velocity);
  
    delay(500);
   
    // Send a MIDI Note Off message to turn it off again
    midi.sendNoteOff(note, velocity);
    delay(500);
    
    midi_ble.update();
    midi.update();
}
```

tttapa, 2025-01-13:
I don't use PlatformIO myself, so I'm afraid I won't be able to provide detailed support.

Control Surface uses the built-in `USB` library from the ESP32 Arduino Core. Could you please test https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/MIDI/MidiController/MidiController.ino and report back? If that also doesn't work, it's probably a problem with PlatformIO rather than Control Surface.

madskjeldgaard, 2025-01-20:
I get a similar problem with Raspberry Pi Pico in Platformio. Is it the same? The problem seems to be that MIDIUSB doesn't detect the architecture. 

```
In file included from .pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:17:
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:18:2: error: #error MIDIUSB can only be used with an USB MCU.
   18 | #error MIDIUSB can only be used with an USB MCU.
      |  ^~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:78:2: error: #error "Unsupported architecture"
   78 | #error "Unsupported architecture"
      |  ^~~~~
Compiling .pio/build/raspberrypi-pico/lib333/Control Surface/MIDI_Constants/MCUNameFromNoteNumber.cpp.o
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:129:9: error: 'EndpointDescriptor' does not name a type
  129 |         EndpointDescriptor len;         // 9
      |         ^~~~~~~~~~~~~~~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:158:9: error: 'IADDescriptor' does not name a type; did you mean 'MIDI_EPDescriptor'?
  158 |         IADDescriptor                      iad;
      |         ^~~~~~~~~~~~~
      |         MIDI_EPDescriptor
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:160:9: error: 'InterfaceDescriptor' does not name a type; did you mean 'MIDI_ASInterfaceDes
criptor'?
  160 |         InterfaceDescriptor                Audio_ControlInterface;
      |         ^~~~~~~~~~~~~~~~~~~
      |         MIDI_ASInterfaceDescriptor
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:164:9: error: 'InterfaceDescriptor' does not name a type; did you mean 'MIDI_ASInterfaceDes
criptor'?
  164 |         InterfaceDescriptor                Audio_StreamInterface;
      |         ^~~~~~~~~~~~~~~~~~~
      |         MIDI_ASInterfaceDescriptor
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:217:9: error: 'EPTYPE_DESCRIPTOR_SIZE' does not name a type
  217 |         EPTYPE_DESCRIPTOR_SIZE epType[2];   ///< Container that defines the two bulk MIDI IN/OUT endpoints types
      |         ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:30:37: error: 'MIDI_BUFFER_SIZE' was not declared in this scope
   30 |         midiEventPacket_t midiEvent[MIDI_BUFFER_SIZE];
      |                                     ^~~~~~~~~~~~~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:35:51: error: too many initializers for 'ring_bufferMIDI'
   35 | ring_bufferMIDI midi_rx_buffer = {{0,0,0,0 }, 0, 0};
      |                                                   ^
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp: In member function 'virtual int MIDI_::getInterface(uint8_t*)':
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:44:17: error: 'D_IAD' was not declared in this scope
   44 |                 D_IAD(MIDI_AC_INTERFACE, 2, MIDI_AUDIO, MIDI_AUDIO_CONTROL, 0),
      |                 ^~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:45:17: error: 'D_INTERFACE' was not declared in this scope; did you mean 'D_AC_INTERFACE'
?
   45 |                 D_INTERFACE(MIDI_AC_INTERFACE,0,MIDI_AUDIO,MIDI_AUDIO_CONTROL,0),
      |                 ^~~~~~~~~~~
      |                 D_AC_INTERFACE
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:53:32: error: 'USB_ENDPOINT_OUT' was not declared in this scope; did you mean 'MIDI_ENDPO
INT_OUT'?
   53 |                 D_MIDI_JACK_EP(USB_ENDPOINT_OUT(MIDI_ENDPOINT_OUT),USB_ENDPOINT_TYPE_BULK,MIDI_BUFFER_SIZE),
      |                                ^~~~~~~~~~~~~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:191:17: note: in definition of macro 'D_MIDI_JACK_EP'
  191 |         { 9, 5, _addr,_attr,_packetSize, 0, 0, 0}
      |                 ^~~~~
Compiling .pio/build/raspberrypi-pico/lib333/Control Surface/MIDI_Inputs/LEDs/NoteCCKPRangeFastLED.cpp.o
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:53:68: error: 'USB_ENDPOINT_TYPE_BULK' was not declared in this scope
   53 |                 D_MIDI_JACK_EP(USB_ENDPOINT_OUT(MIDI_ENDPOINT_OUT),USB_ENDPOINT_TYPE_BULK,MIDI_BUFFER_SIZE),
      |                                                                    ^~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:191:23: note: in definition of macro 'D_MIDI_JACK_EP'
  191 |         { 9, 5, _addr,_attr,_packetSize, 0, 0, 0}
      |                       ^~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:53:91: error: 'MIDI_BUFFER_SIZE' was not declared in this scope
   53 |                 D_MIDI_JACK_EP(USB_ENDPOINT_OUT(MIDI_ENDPOINT_OUT),USB_ENDPOINT_TYPE_BULK,MIDI_BUFFER_SIZE),
      |                                                                                           ^~~~~~~~~~~~~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.h:191:29: note: in definition of macro 'D_MIDI_JACK_EP'
  191 |         { 9, 5, _addr,_attr,_packetSize, 0, 0, 0}
      |                             ^~~~~~~~~~~
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp:55:32: error: 'USB_ENDPOINT_IN' was not declared in this scope; did you mean 'MIDI_ENDPOI
NT_IN'?
   55 |                 D_MIDI_JACK_EP(USB_ENDPOINT_IN(MIDI_ENDPOINT_IN),USB_ENDPOINT_TYPE_BULK,MIDI_BUFFER_SIZE),
      |                                ^~~~~~~~~~~~~~~
[... 28 lines cut ...]
Compiling .pio/build/raspberrypi-pico/lib333/Control Surface/MIDI_Interfaces/BLEMIDI/BLEMIDIPacketBuilder.cpp.o
.pio/libdeps/raspberrypi-pico/MIDIUSB/src/MIDIUSB.cpp: In member function 'size_t MIDI_::write(const uint8_t*, size_t)':
[... cut ...]
  193 |         epType[1] = EP_TYPE_BULK_IN_MIDI;               // MIDI_ENDPOINT_IN
      |                     ^~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/raspberrypi-pico/lib333/Control Surface/MIDI_Interfaces/BLEMIDI/ESP32/ble2902.c.o
Compiling .pio/build/raspberrypi-pico/lib333/Control Surface/MIDI_Interfaces/BLEMIDI/ESP32/esp_enums2string.c.o
*** [.pio/build/raspberrypi-pico/libfc1/MIDIUSB/MIDIUSB.cpp.o] Error 1
========================================================= [FAILED] Took 1.71 seconds =========================================================

```

tttapa, 2025-01-20:
@madskjeldgaard that's a different problem. You need to ignore the `MIDIUSB` library on platforms that don't need it, otherwise, PlatformIO will try to compile it anyway, which of course won't work. See https://tttapa.github.io/Control-Surface/Doxygen/d8/da8/md_pages_Installation.html#platformio.

madskjeldgaard, 2025-01-21:
Ah okay thanks !

tttapa, 2025-01-24:
@ashojon Which version of the arduino-esp32 core are you using? USB MIDI support was added in version 3.0.0: https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha3

ashojon, 2025-01-29:
@tttapa I'm not entirely sure. The platform I use is Espressif 6.9.0

tttapa, 2025-01-29:
That version is based on v2.0.17 of the Arduino ESP32 core, so it doesn't support MIDI over USB yet.  
https://github.com/platformio/platform-espressif32/releases/tag/v6.9.0

ashojon, 2025-01-29:
Does this mean I can't use PlatformIO and Visual Studio Code to develop MIDI over USB for ESP32?

tttapa, 2025-01-29:
I'm afraid so. Here's a thread I found while googling: https://github.com/platformio/platform-espressif32/issues/1225

Regarding VSCode, you can still use it for editing your code, you don't have to use the Arduino IDE for that. (Although I haven't personally used it for ESP32+Arduino, only for AVR boards or directly using the ESP-IDF).

AriESQ, 2025-02-08:
This question has been answered, please close issue.

closed 2025-02-10 by ashojon