tttapa/Control-Surface #1109: Support for Pico 2 board?

madskjeldgaard, 2025-01-13:
Hi! 

I see Raspberry Pi Pico in the list of supported boards and it seems to be RP2040-based ones. But is the Pico 2 supported as well, including it's new mcu rp2350?

Thanks for making this awesome project!!

tttapa, 2025-01-18:
I didn't order any Pico 2's yet, so I can't test it myself, but the Control Surface examples seem to compile without issues (using [**earlephilhower/arduino-pico**](https://github.com/earlephilhower/arduino-pico)).   
At first sight, most of the APIs that Control Surface relies on are the same between the RP2040 and the RP2350, so I don't expect too many issues. Some components to double-check are the encoders and the USB MIDI backends, but most features should work out-of-the box.

If you (or anyone else coming across this thread) own a Pico 2, feel free to try it out and report your findings here!

madskjeldgaard, 2025-01-20:
Thanks a lot! I have a Pico 2 and will test it out. I honestly think everything will work fine. I'll close the issue now but reopen if it doesn't work :)

closed 2025-01-20 by madskjeldgaard

madskjeldgaard, 2026-01-31:
> Hello [@madskjeldgaard](https://github.com/madskjeldgaard) ! Sorry to reopen this issue, but I just wanted to check, were you successful using the Pico 2?
> 
> I've been using a teensy 4.1 for my prototype controller but I wanted to get the costs down, and the Pico 2 looks like a good option. Did you run into any issues or did it all work as expected? Has it been enough for your use cases?
> 
> Thanks for your help!

The pico 2 is more than powerful enough for a controller. But it doesn't have as many ADC's as teensy 4.1, so that's the biggest decision to make between the two :) If you need lots of pots/sliders then teensy is better. (or an external dac for pico2)