tttapa/Control-Surface #1126: PlatformIO build Pico RP2040 addressmap.h:71:21: error: type qualifiers ignored on cast result type

Erriez, 2025-06-18:
**Describe the bug**
PlatformIO build error RP2040:

```
C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/
pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:71:21: 
error: type qualifiers ignored on cast result type [-Werror=ignored-qualifiers]
```

**To Reproduce**
Steps to reproduce the behavior:

- Install VSCode with PlatformIO plugin.
- Create new PlatformIO project with Pico board and Arduino platform. 
- Add library `Control-Surface`.
- Add `#include <Control_Surface.h>` is sufficient to reproduce the build error.
- Build project.
- Output:

```
 *  Executing task in folder pico-usb-midi: C:\Users\user\.platformio\penv\Scripts\platformio.exe run --environment pico 

Processing pico (platform: raspberrypi; board: pico; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.17.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES: 
 - framework-arduino-mbed @ 4.3.1 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Control Surface @ 1.2.0-4+sha.f85f2e3
Building in release mode
Compiling .pio\build\pico\src\main.cpp.o
Archiving .pio\build\pico\lib10a\libSPI.a
Compiling .pio\build\pico\libf0c\Control Surface\Submodules\Encoder\AHEncoder.cpp.o
In file included from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/pico_platform/include/pico/platform.h:19,
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_base/include/pico.h:22,
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/objects.h:30,       
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/device.h:35,        
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/platform/include/platform/platform.h:28,
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/platform/include/platform/FileHandle.h:25,
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/macros.h:41,
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO/pins_arduino.h:2,
                 from C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/Arduino.h:76,
                 from .pio\libdeps\pico\Control Surface\src/AH/Math/FixArduinoMacros.hpp:6,
                 from .pio\libdeps\pico\Control Surface\src/AH/Math/MinMaxFix.hpp:3,
                 from .pio\libdeps\pico\Control Surface\src/AH/Arduino-Wrapper.h:3,
                 from .pio\libdeps\pico\Control Surface\src\Submodules\Encoder\AtomicPosition.hpp:3,
                 from .pio\libdeps\pico\Control Surface\src\Submodules\Encoder\AHEncoder.hpp:3,
                 from .pio\libdeps\pico\Control Surface\src\Submodules\Encoder\AHEncoder.cpp:3:
.pio\libdeps\pico\Control Surface\src\Submodules\Encoder\DirectPinRead.hpp: In function 'cs::DirectPinRead cs::direct_pin_read(ArduinoPin_t)':
C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:71:21: error: type qualifiers ignored on cast result type [-Werror=ignored-qualifiers]
   71 | #define SIO_BASE _u(0xd0000000)
      |                     ^~~~~~~~~~
C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/platform_defs.h:16:15: note: in definition of macro '_u'
   16 | #define _u(x) x ## u
      |               ^
C:\Users\user\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/sio.h:174:34: note: in expansion of macro 'SIO_BASE'
  174 | #define sio_hw ((sio_hw_t *const)SIO_BASE)
      |                                  ^~~~~~~~
.pio\libdeps\pico\Control Surface\src\Submodules\Encoder\DirectPinRead.hpp:83:14: note: in expansion of macro 'sio_hw'
   83 |     return {&sio_hw->gpio_in, uint32_t(1) << pin};
[... 54 lines cut ...]
void loop() {
  midi.sendNoteOn(address, velocity);
  delay(500);
  midi.sendNoteOff(address, velocity);
  delay(500);

  midi.update(); // Handle or discard MIDI input
}
```

platformio.ini:
```ini
[env:pico]
platform = raspberrypi
board = pico
framework = arduino

[env]
lib_deps = https://github.com/tttapa/Control-Surface.git#main
lib_ignore = MIDIUSB
```

Same project works fine in the Arduino IDE 2.3.6 Board Arduino Mbed OS RP2040, Control-Surface library 2.0.0.

Any advice?

tttapa, 2025-06-18:
This is a problem in the Pico SDK. It was fixed almost four years ago, so you seem to be using a very outdated version:
https://github.com/raspberrypi/pico-sdk/commit/0562089faba91b3996f5f8e479489fb03bd74922#diff-f5f288aee832b50080097b00c961d972caff9e421228b9ecf5e1f5228e81807f

Please try updating your PlatformIO boards/platforms.

Erriez, 2025-06-21:
Thanks for your quick response. In short: Control-Surface with MIDI on native USB port works on Pico RP2040, thanks to your great description:

> I'm afraid I can't really provide detailed support for PlatformIO, since I've never used the configuration you're describing.

Good to know this. I switched to Arduino IDE v2 instead. This is what I've done:

`Raspberry Pi Pico RP2040` board with `ControlSurface.h` and `USBMIDI_Interface`:
- `Arduino Mbed OS`:
  - `Ardunio IDE 2.3.6`:
    - Board Manager: Install `Arduino Mbed OS RP2040 Boards` `v4.3.1` -> Currently supports `RP2040` board only, not `RP2350`
	- Tools | Board: `Raspberry Pi Pico`
      - MIDI on native USB device port works
- `earlephilhower`:
  - `Ardunio IDE 2.3.6`:
    - File | Preferences | Additional boards manager URLs: https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json
	- Board Manager: Install `Raspberry Pi Pico/RP204/RP2350` `v4.5.4` -> Supports `RP2040` and `RP2350` boards
    - Tools | Board: `Raspberry Pi Pico`
	  - [MIDI on serial port as fallback](https://tttapa.github.io/Control-Surface/Doxygen/de/d05/classUSBMIDI__Interface.html) and no MIDI on USB device port, because `TinyUSB` is not enabled by default.
	- Library Manager: Install `Adafruit TinyUSB Library` `v3.7.0` (without dependencies)
	  - Without `Adafruit TinyUSB Library`, still MIDI on serial port
	- Tools | USB Stack: `Adafruit TinyUSB`
      - MIDI on native USB device port works

`Arduino Mbed OS` aka `Mbed OS core` summary as I understand correctly:
- A real-time operating system (RTOS) developed by ARM.
- It’s designed for 32-bit ARM Cortex-M microcontrollers.
- It supports the Raspberry Pi Pico RP2040 Cortex-M0+ Board.
- At this moment, it does not officially support the Raspberry Pi Pico2 RP2350 Cortex-M33 board.
- Raspberry Pico2 RP2350 board is currently supported by the `Raspberry Pi Pico Arduino core` developed by `Earle F. Philhower` (aka Philhower core) as alternative to the `Mbed OS core`

I'll try to port it to PlatformIO later and will post the results when it up and running.

tttapa, 2025-06-21:
Thanks for the detailed instructions!

Some quick comments: 
- The Adafruit TinyUSB library is included in the earlephilhower arduino-pico core, so installing it through the library manager is not necessary (and probably won't have any effect, since the built-in one will be used).
- Mbed OS is end-of-life, so it's not really worth looking into at this point: https://blog.arduino.cc/2024/07/24/the-end-of-mbed-marks-a-new-beginning-for-arduino/

Erriez, 2025-06-21:
PlatformIO debugging with two Raspberry Pi Pico's RP2040 over SWD also up and running. :-)

1. See [getting-started-with-pico.pdf](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf):
    - Section Appendix A: Debugprobe | Debug with a second Pico or Pico 2
    - See Figure 9: Wiring schematic two Pico RP2040 boards over SWD and optional UART (hardware serial for monitor): One board is the debugger with USB connected, second is the target board to program/debug.
2. Download `debugprobe_on_pico.uf2` or `debugprobe_on_pico2.uf2` from: https://github.com/raspberrypi/debugprobe/releases (tested release `v2.2.2`).
3. Boot/power-on the debugger Pico (2) with the `BOOTSEL` button pressed and copy the downloaded `.uf2` file onto the drive letter.
4. Add lines below to `platformio.ini`:

```ini
debug_tool = picoprobe
upload_protocol = picoprobe
debug_init_break = tbreak setup
```

Click in the left bar `Run and Debug`, select the project and click the play button. Wait until breakpoint is reached in function `setup`.

[image]

This is the cheapest debugging setup ever. 🥇