tttapa/Control-Surface #1133: Example code throws error: "'BluetoothMIDI_Interface' does not name a type"

romandesign, 2025-08-08:
**Describe the bug**
When I copy-paste the BLEMIDI-Adapter.ino code into Arduino IDE, after installing the library, I get an error "'BluetoothMIDI_Interface' does not name a type"

**To Reproduce**
Istall library via library manager or manually - same effect. I'm using ESP8266 board, I tried selected Generic and NodeMCU, but it's the same error.

**Expected behavior**
I expect an example to compile and run.

**Code**
the exact code from the example: [BLEMIDI-Adapter.ino](https://github.com/tttapa/Control-Surface/blob/main/examples/3.%20MIDI%20Interfaces/BLEMIDI-Adapter/BLEMIDI-Adapter.ino)

tttapa, 2025-08-08:
Hi,
The ESP8266 does not support Bluetooth. See <https://tttapa.github.io/Control-Surface/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html> for a list of supported devices.

tttapa, 2025-08-08:
> but it seems to hang when plugged into Roland GoKeys5

Which port did you plug the Pro Micro into? The _USB memory_ port does not seem to support MIDI, and the _USB Computer Port_ is a USB device port, so you cannot connect it to the Arduino (which is another USB device, you need at least one USB host).

romandesign, 2025-08-08:
I plugged it into the wifi dongle USB A port, which has an undocumented MIDI support. When I plug  Alesis Vortex 2 via MIDI cable to the same port, it powers up from the port, and all sliders and notes work controlling the Roland, and I think Alesis keytar is also a client and not a host, because it doesn't receive any MIDI, it's a controller. It has a USB A plug goign into USB A on Roland. Roland GoKeys 5 receives all data on channel 4 and then it just works (when keytar is programmed to send on ch4). When I plug Pro Micto into my PC I can see MIDI controllers working and can control a soft synth. But Roland doesn't respond.
I'm not sure if I can test if Alesis keytar is a host, but I hoghly doubt it. So Roland seems to be working as a host, just not with the Pro Micro for some reason.

tttapa, 2025-08-08:
The Alesis is indeed a USB device and not a host (but this is independent of whether it sends or receives MIDI).

If the MIDI support of the Roland's WiFi dongle port is undocumented, it's going to be tough to debug, I'm afraid.

romandesign, 2025-08-08:
I just don't understand what's different. If Alesis is a MIDI USB device and Pro Micro is a device, and same CCc and notes work from Alesis, why Pro Micro doesn't work? I tried 3 MIDI libraries before this one which seems to be the best suited for midi controllers. There's one symptom i can see: when i hooked up my original controller with it's own i2c display to Roland, the display showing affected CC values, the display refresh slowed down drastically compared to when connecting to PC, and CC values would jump in large increments instead of being smooth. When i tried a simple sketch with an empty Arduino, just a single pot connected and built-in LED that supposed to flash when transmitting midi, the LED would flash when connected to PC any time i turn the pot and send CC data. But with Roland it would only block once and then nothing. Sleeping makes the Arduino hang i think. I thought maybe there's not enough power coming from the Roland but using external power did not help at all, and Alesis is powered just fine with it's many LEDs.

tttapa, 2025-08-14:
> But with Roland it would only block once and then nothing. Something makes the Arduino hang i think.

The hanging is likely caused by the Roland not accepting any of the USB transfers (most likely because it is not enumerated correctly by the Roland's USB implementation). Once the buffer on the Arduino side is full, any further transfers will block (and timeout).

For future reference:
- https://forum.arduino.cc/t/arduino-pro-micro-midi-controller-not-recognized-by-roland-synth-but-recognized-by-pc/1400971?u=pieterp
- https://forum.arduino.cc/t/usb-midi-on-pro-micro-freezes-my-roland-synth-need-advice/1401496?u=pieterp

tttapa, 2025-08-15:
Thanks for trying it out! I should have some time next week to look into the error you're facing.

uecoding, 2025-08-21:
> Fixed in [version 2.1.0](https://github.com/tttapa/Control-Surface/releases/tag/2.1.0).

That worked great! Thank you so much. 

Is there an example sketch where I can use either bluetooth or usb depending on a switch reading on startup? I tried putting USBMIDI_Interface and BluetoothMIDI_Interface inside an if condition in setup function but no luck.

tttapa, 2025-08-22:
If you declare them inside of an if statement, the MIDI interfaces get destroyed at the end of the block.

There are many ways to achieve what you want: Here's one where both interfaces are created and initialized, but only one is used by `Control_Surface`.

```cpp
#include <Control_Surface.h>

BluetoothMIDI_Interface midi_ble;
USBMIDI_Interface midi_usb;

pin_t ble_usb_mode_switch_pin = 4;

void setup() {
  pinMode(ble_usb_mode_switch_pin, INPUT_PULLUP);
  delay(1);
  if (digitalRead(ble_usb_mode_switch_pin) == HIGH)
    midi_ble.setAsDefault();
  else
    midi_usb.setAsDefault();
  // Initialize the Control Surface (after changing the default)
  Control_Surface.begin();
}

void loop() {
  static uint8_t counter = 0;
  static AH::Timer<millis> timer{500};
  if (timer)
    Control_Surface.sendControlChange(0, (++counter) & 0x7F);
  Control_Surface.loop();
}
```

With the code above, you could still use both interfaces directly if you wanted to (by calling `midi_ble.send()` or `midi_usb.send()`, for example). If you don't need both interfaces to be functional at the same time, you can disable one of them before calling `Control_Surface.begin()`:
```cpp
void setup() {
  pinMode(ble_usb_mode_switch_pin, INPUT_PULLUP);
  delay(1);
  if (digitalRead(ble_usb_mode_switch_pin) == HIGH)
    midi_usb.disable();
  else
    midi_ble.disable();
  // Initialize the Control Surface (after disabling the unused one)
  Control_Surface.begin();
}
```

To avoid having to keep both interfaces in memory at the same time, you can use a `variant` type:

```cpp
#include <Control_Surface.h>
#include <variant>

std::variant<BluetoothMIDI_Interface, USBMIDI_Interface> midi;

pin_t ble_usb_mode_switch_pin = 4;

void setup() {
  pinMode(ble_usb_mode_switch_pin, INPUT_PULLUP);
  delay(1);
  if (digitalRead(ble_usb_mode_switch_pin) == HIGH)
    midi.emplace<0>();
  else
    midi.emplace<1>();
  // Initialize the Control Surface (after constructing the interface)
  Control_Surface.begin();
}

void loop() {
  static uint8_t counter = 0;
  static AH::Timer<millis> timer{500};
  if (timer)
    Control_Surface.sendControlChange(0, (++counter) & 0x7F);
  Control_Surface.loop();
}
```